f4exb/sdrangel #185: Ubuntu 16.04LTE build

siralexbox, 2018-06-06:
The build for Ubuntu 16.04LTE sdrangel_3.14.7-1_amd64_xenial.deb is not installing.

(Reading database ... 306468 files and directories currently installed.)
Preparing to unpack sdrangel_3.14.7-1_amd64_xenial.deb ...
Unpacking sdrangel (3.14.7-1) over (3.14.7-1) ...
dpkg: dependency problems prevent configuration of sdrangel:
 sdrangel depends on libnanomsg0 (>= 0.4~beta); however:
  Package libnanomsg0 is not installed.
 sdrangel depends on libopencv-dev; however:
  Package libopencv-dev is not installed.
 sdrangel depends on libavcodec-dev; however:
  Package libavcodec-dev is not installed.
 sdrangel depends on libavformat-dev; however:
  Package libavformat-dev is not installed.

dpkg: error processing package sdrangel (--install):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.23-0ubuntu10) ...
Errors were encountered while processing:
 sdrangel

f4exb, 2018-06-07:
See: https://github.com/f4exb/sdrangel#debian-distributions
`sudo apt-get -f install` should fix the issue

siralexbox, 2018-06-10:
Thanks but I'm coming from there.
Downloaded sdrangel_3.14.7-1_amd64_xenial.deb
sudo dpkg -i sdrangel_3.14.7-1_amd64_xenial.deb
And get error:
Selecting previously unselected package sdrangel.
(Reading database ... 399513 files and directories currently installed.)
Preparing to unpack sdrangel_3.14.7-1_amd64_xenial.deb ...
Unpacking sdrangel (3.14.7-1) ...
dpkg: dependency problems prevent configuration of sdrangel:
 sdrangel depends on libnanomsg0 (>= 0.4~beta); however:
  Package libnanomsg0 is not installed.
 sdrangel depends on libopencv-dev; however:
  Package libopencv-dev is not installed.

dpkg: error processing package sdrangel (--install):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.23-0ubuntu10) ...
Errors were encountered while processing:
 sdrangel

And when I try to install libopencv-dev also get an error

Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies:
 libopencv-dev : Depends: libopencv-core-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-ml-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-imgproc-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-video-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-objdetect-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-highgui-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-calib3d-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-flann-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-features2d-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-legacy-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-contrib-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-ts-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-photo-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-videostab-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-stitching-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-gpu-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-superres-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv-ocl-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv2.4-java (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libopencv2.4-jni (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libcv-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libhighgui-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
                 Depends: libcvaux-dev (= 2.4.9.1+dfsg-1.5ubuntu1) but it is not going to be installed
 sdrangel : Depends: libnanomsg0 (>= 0.4~beta) but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).

Even if using synaptic the package libopencv-dev it broken and showing red.
If I run sudo apt-get -f install

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following packages were automatically installed and are no longer required:
  libasound2-dev libportaudiocpp0 uuid-dev
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  sdrangel
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 24.3 MB disk space will be freed.
Do you want to continue? [Y/n]

f4exb, 2018-06-11:
Did you do the usual: 
```
sudo apt-get update
sudo apt-get upgrade
```

siralexbox, 2018-06-12:
Here is my /etc/apt/sources.list what is missing ?

# deb cdrom:[Ubuntu-GNOME 16.04.3 LTS _Xenial Xerus_ - Release amd64 (20170801)]/ xenial main multiverse restricted universe

deb http://us.archive.ubuntu.com/ubuntu/ xenial main restricted
deb http://us.archive.ubuntu.com/ubuntu/ xenial-updates main restricted
deb http://us.archive.ubuntu.com/ubuntu/ xenial universe
deb http://us.archive.ubuntu.com/ubuntu/ xenial-updates universe
deb http://us.archive.ubuntu.com/ubuntu/ xenial multiverse
deb http://us.archive.ubuntu.com/ubuntu/ xenial-updates multiverse
deb http://us.archive.ubuntu.com/ubuntu/ xenial-backports main restricted universe multiverse
deb http://archive.canonical.com/ubuntu xenial partner
deb http://security.ubuntu.com/ubuntu xenial-security main restricted
deb http://security.ubuntu.com/ubuntu xenial-security universe
deb http://security.ubuntu.com/ubuntu xenial-security multiverse
# Add by me
deb http://ppa.launchpad.net/bladerf/bladerf/ubuntu xenial main
deb http://ppa.launchpad.net/danielrichter2007/grub-customizer/ubuntu xenial main
deb http://ppa.launchpad.net/gezakovacs/ppa/ubuntu xenial main
deb [arch=amd64] http://dl.google.com/linux/chrome/deb/ stable main
deb http://ppa.launchpad.net/gpredict-team/ppa/ubuntu xenial main
deb http://ppa.launchpad.net/gqrx/gqrx-sdr/ubuntu xenial main
deb http://ppa.launchpad.net/kamalmostafa/fldigi/ubuntu xenial main
deb http://download.mono-project.com/repo/debian xenial main
deb http://ppa.launchpad.net/myriadrf/drivers/ubuntu xenial main
deb http://ppa.launchpad.net/myriadrf/gnuradio/ubuntu xenial main
deb http://ppa.launchpad.net/nilarimogard/webupd8/ubuntu xenial main
deb http://ppa.launchpad.net/paulo-miguel-dias/mesa/ubuntu xenial main
deb http://repository.spotify.com stable non-free
deb [arch=amd64,i386] http://repo.steampowered.com/steam/ precise steam
deb http://ppa.launchpad.net/teejee2008/ppa/ubuntu xenial main
deb [arch=amd64] http://packages.microsoft.com/repos/vscode stable main
deb http://ppa.launchpad.net/yunnxx/gnome3/ubuntu xenial main

siralexbox, 2018-06-18:
OK I got it to compile.
Ditched 16.04 LTE and updated to 18.04 LTE, all good now.
Thanks for the support.

f4exb, 2018-06-19:
I am now currently developing on 18.04 LTE so I am less likely to spot issues on 16.04 although I have one box and the build system still running on it.
Closing now.

closed 2018-06-19 by f4exb