f4exb/sdrangel #186: dsdcc

KC7NOA, 2018-06-13:
hi not sure where to put...

successful build of dsdcc but sdrangel still not detect it ...

sudo ldconfig tried

f4exb, 2018-06-13:
Not sure it can detect it automatically even if you install dsdcc in the standard paths. To be sure you can always specify the dsdcc paths on the cmake command line. See: https://github.com/f4exb/sdrangel#dsd-digital-speech-decoder

KC7NOA, 2018-06-13:
Ok .. tried without it for now ...

Tried to build and hit error on pluto ... i dont have one nor should be tried to build.

KC7NOA, 2018-06-13:
``Scanning dependencies of target plutosdrdevice
[ 48%] Building CXX object devices/plutosdr/CMakeFiles/plutosdrdevice.dir/deviceplutosdr.cpp.o
[ 48%] Building CXX object devices/plutosdr/CMakeFiles/plutosdrdevice.dir/deviceplutosdrbox.cpp.o
/home/kc7noaa/Downloads/sdrangel/devices/plutosdr/deviceplutosdrbox.cpp: In constructor ‘DevicePlutoSDRBox::DevicePlutoSDRBox(const string&)’:
/home/kc7noaa/Downloads/sdrangel/devices/plutosdr/deviceplutosdrbox.cpp:50:52: error: ‘iio_create_context_from_uri’ was not declared in this scope
     m_ctx = iio_create_context_from_uri(uri.c_str());
                                                    ^
/home/kc7noaa/Downloads/sdrangel/devices/plutosdr/deviceplutosdrbox.cpp: In static member function ‘static bool DevicePlutoSDRBox::probeURI(const string&)’:
/home/kc7noaa/Downloads/sdrangel/devices/plutosdr/deviceplutosdrbox.cpp:89:50: error: ‘iio_create_context_from_uri’ was not declared in this scope
     ctx = iio_create_context_from_uri(uri.c_str());
                                                  ^
/home/kc7noaa/Downloads/sdrangel/devices/plutosdr/deviceplutosdrbox.cpp: In member function ‘bool DevicePlutoSDRBox::openRx()’:
/home/kc7noaa/Downloads/sdrangel/devices/plutosdr/deviceplutosdrbox.cpp:272:21: error: ‘const struct iio_data_format’ has no member named ‘repeat’
                 df->repeat);
                     ^
/home/kc7noaa/Downloads/sdrangel/devices/plutosdr/deviceplutosdrbox.cpp: In member function ‘bool DevicePlutoSDRBox::openTx()’:
/home/kc7noaa/Downloads/sdrangel/devices/plutosdr/deviceplutosdrbox.cpp:299:21: error: ‘const struct iio_data_format’ has no member named ‘repeat’
                 df->repeat);
                     ^
/home/kc7noaa/Downloads/sdrangel/devices/plutosdr/deviceplutosdrbox.cpp:320:21: error: ‘const struct iio_data_format’ has no member named ‘repeat’
                 df->repeat);
                     ^
devices/plutosdr/CMakeFiles/plutosdrdevice.dir/build.make:86: recipe for target 'devices/plutosdr/CMakeFiles/plutosdrdevice.dir/deviceplutosdrbox.cpp.o' failed
make[2]: *** [devices/plutosdr/CMakeFiles/plutosdrdevice.dir/deviceplutosdrbox.cpp.o] Error 1
CMakeFiles/Makefile2:1219: recipe for target 'devices/plutosdr/CMakeFiles/plutosdrdevice.dir/all' failed
make[1]: *** [devices/plutosdr/CMakeFiles/plutosdrdevice.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
kc7noaa@odroid64:~/Downloads/sdrangel/build$ 

kc7noaa@odroid64:~/Downloads/sdrangel/build$ cmake ../
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /usr/lib/aarch64-linux-gnu/libGL.so  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Boost version: 1.58.0
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.4
-- Found FFTW3F: /usr/lib/aarch64-linux-gnu/libfftw3f.so  
-- Found FFTW3F: /usr/include, /usr/lib/aarch64-linux-gnu/libfftw3f.so, /usr/lib/aarch64-linux-gnu/libfftw3f_threads.so
-- Checking for module 'libdsdcc'
--   No package 'libdsdcc' found
-- libdsdcc not found.
-- Architecture: aarch64
-- Aarch64 always has NEON SIMD instructions
-- Compiling for 16 bit Rx DSP chain
-- Compiling with SIMD instructions for DSP if available
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.20
-- Found libusb-1.0: /usr/include/libusb-1.0, /usr/lib/aarch64-linux-gnu/libusb-1.0.so
-- Checking for module 'libbladeRF'
--   No package 'libbladeRF' found
-- libbladerf not found.
-- Checking for module 'libhackrf'
--   No package 'libhackrf' found
-- libhackrf not found.
-- Found Lime Suite: /usr/local/include, /usr/local/lib/libLimeSuite.so
-- Checking for module 'libiio'
[... 607 lines cut ...]
/home/kc7noaa/Downloads/sdrangel/devices/plutosdr/deviceplutosdrbox.cpp:50:52: error: ‘iio_create_context_from_uri’ was not declared in this scope
     m_ctx = iio_create_context_from_uri(uri.c_str());
                                                    ^
/home/kc7noaa/Downloads/sdrangel/devices/plutosdr/deviceplutosdrbox.cpp: In static member function ‘static bool DevicePlutoSDRBox::probeURI(const string&)’:
/home/kc7noaa/Downloads/sdrangel/devices/plutosdr/deviceplutosdrbox.cpp:89:50: error: ‘iio_create_context_from_uri’ was not declared in this scope
     ctx = iio_create_context_from_uri(uri.c_str());
                                                  ^
/home/kc7noaa/Downloads/sdrangel/devices/plutosdr/deviceplutosdrbox.cpp: In member function ‘bool DevicePlutoSDRBox::openRx()’:
/home/kc7noaa/Downloads/sdrangel/devices/plutosdr/deviceplutosdrbox.cpp:272:21: error: ‘const struct iio_data_format’ has no member named ‘repeat’
                 df->repeat);
                     ^
/home/kc7noaa/Downloads/sdrangel/devices/plutosdr/deviceplutosdrbox.cpp: In member function ‘bool DevicePlutoSDRBox::openTx()’:
[... cut ...]
devices/plutosdr/CMakeFiles/plutosdrdevice.dir/build.make:86: recipe for target 'devices/plutosdr/CMakeFiles/plutosdrdevice.dir/deviceplutosdrbox.cpp.o' failed
make[2]: *** [devices/plutosdr/CMakeFiles/plutosdrdevice.dir/deviceplutosdrbox.cpp.o] Error 1
CMakeFiles/Makefile2:1219: recipe for target 'devices/plutosdr/CMakeFiles/plutosdrdevice.dir/all' failed
make[1]: *** [devices/plutosdr/CMakeFiles/plutosdrdevice.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
kc7noaa@odroid64:~/Downloads/sdrangel/build$

KC7NOA, 2018-06-13:
sorry ... whole build till error ....im in a rush to get to work

unixpunk, 2018-06-13:
This might be because cmake sees libiio installed but not also libad9361-iio?  So either install that lib or try uninstall libiio.  If neither work, then maybe @f4exb has a flag you can pass to cmake to disable libiio/plutosdr support altogether.  For the dsd libs, you need to tell cmake where these are as the previous post mentioned.  I might also suggest you start fresh, from the build dir: make clean && rm -rf * and then use cmake again.  HTH

KC7NOA, 2018-06-13:
I made a new folder an git pulled a fresh ... that was my output

The refering post was right to the issues main page and not something specific

KC7NOA, 2018-06-13:
I think libiio was built from git trying to satisfy some other dependencies with another program. I'm not sure where to get the most current from, it's not in the Repository

unixpunk, 2018-06-13:
github has it, that's where I saw the latest is .15.  Its owned by analogdevices.  Sorry I can't help further.  Either uninstall it, upgrade it or install the libad9361-iio portion to satisfy the pluto requirements of sdrangel.  Again, just guessing, never had this problem but I have the latest libiio and libad9361-iio because I do have a PlutoSDR.

f4exb, 2018-06-23:
Works fine with latest libiio (0.15). Next release to be compiled with this one.

KC7NOA, 2018-06-23:
I couldnt get .15 to build

I ended up just removing both packages

f4exb, 2018-06-28:
v4.0.1 built with latest libbiio (0.15)

closed 2018-06-28 by f4exb