f4exb/sdrangel #187: display

KC7NOA, 2018-06-17:
MESA-LOADER: failed to retrieve device information
libGL error: unable to load driver: mali_drm_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: mali_drm

is that an issue?

KC7NOA, 2018-06-17:
is it hard to support GLESv1 GLESv2 and EGL?

KC7NOA, 2018-06-20:
Odroid-c2   quad core mali-450 

SdrAngel runs well with LimeSDR with a sample rate of 5Ms/ps ... software decimation at 2 and hardware decimation at 1 ....

Just see this when starting and was wondering if there was any way to improve performance on my C2

f4exb, 2018-06-21:
To improve the performance of the whole configuration you should increase the hardware decimation and reduce the host sample rate accordingly. Hardware decimation capability is one of the big advantages of LimeSDR.

closed 2018-06-28 by f4exb

KC7NOA, 2018-06-29:
Works best other way around ... i can get 15M/s with hardware at 1 and software at 4 ... sample around 7M on usb2.0

Closed #187<https://github.com/f4exb/sdrangel/issues/187>.