f4exb/sdrangel #188: SDRangel and Tx HackRF NFM freeze

siralexbox, 2018-06-21:
Ubunto 18.04 LTS
When using SDRangel with HackRF to transmit in NFM if audio is not muted crashes SDRangel.
If I have anything already running that used the capture audio source SDRangel freezes.

closed 2018-06-28 by f4exb

siralexbox, 2018-07-15:
Sorry for my late reply SR is 2400k
Ive made a video showing.
I notice that even if Im capturing sound for the screen cast SDRangel freezes, seems that is anything used the sound channels before SDRandel, its freezes.

[https://youtu.be/X3LeyElzJQk](url)

f4exb, 2018-07-15:
I see one issue there: you are not interpolating at all to get to 2400k. This is a huge step for the channelizer to go from 48k which is the baseband for the NFM modulator to 2400k. The interpolator in the channelizer is not designed for large interpolation but the final interpolator from baseband to the stream sent to the HackRF is. Can you please set the interpolation factor to 16 (150k) or 32 (75k)?