f4exb/sdrangel #190: not sdrangel

KC7NOA, 2018-06-27:
ineed some help fixing this ...

i think i had an error installing qt5 ...

CMake Error at /usr/lib/aarch64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfig.cmake:27 (message):
  The imported target "Qt5::Gui" references the file

     "/usr/lib/aarch64-linux-gnu/libEGL.so"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/lib/aarch64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/lib/aarch64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake:32 (_qt5_Gui_check_file_exists)
  /usr/lib/aarch64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake:54 (_qt5gui_find_extra_libs)
  /usr/lib/aarch64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfig.cmake:184 (include)
  /usr/lib/aarch64-linux-gnu/cmake/Qt5Widgets/Qt5WidgetsConfig.cmake:101 (find_package)
  CMakeLists.txt:57 (find_package)

f4exb, 2018-06-27:
Can you tell us what is the issue with SDRangel? If this is only a Qt5 installation issue I will close this ticket.

KC7NOA, 2018-06-28:
Im not 100% sure ... Qt5 is there/installed but part cant be found. .. i tried making a link from where the file was thought to be and to where i found it by searching ... to no avail

KC7NOA, 2018-06-30:
ok, fresh install of Ubuntu 18 and ....again the same issue ....

i think something in the Ubuntu 16.04 newer version packages break odroid-c2 mali-x11 video stuff ...

Sid Boyce suggested a sudo apt-get install --reinstall mali-x11 

and seems to be working ....

though when starting sdrangel ,,,.... there is an excessive amount of time before the gui come up ....

f4exb, 2018-07-02:
If OpenGL does not work correctly on your machine it is quite possible that the OpenGL parts simply do not display while the rest is working. 

Concerning the startup time it all depends on what you call an "excessive amount of time". You have to consider that it needs to fetch data from the connected devices. Some can take some time for example LimeSDR and PlutoSDR are not very quick. It can also depend on how the USB system is responding.

KC7NOA, 2018-07-02:
Its just way longer compared to tge same hardware runn8ng Ubuntu 16.04

f4exb, 2018-07-03:
No such issue here with Ubuntu 18.04 and Intel CPU. It must be something with your particular hardware.

KC7NOA, 2018-07-04:
software for my hardware ... Ubuntu 16 was fine ....

f4exb, 2018-07-04:
Nothing to be done on my part sorry.

closed 2018-07-04 by f4exb

KC7NOA, 2018-07-04:
Yep .... turns out the MMap in the kernel was the problem .... Odroid had a problem with Chrome right click  
     locking up chrome ... they fixed that and released a new kernel ... it seems to of been the same issue here ....

YEA!!!!

KC7NOA, 2018-07-04:
Wanted to add that Quad core usage is about 1/2 that of gqrx ....

GOOD SHOW MAN!!!!!

f4exb, 2018-07-05:
Appreciate the report about performance vs gqrx. I care particularly about performance when writing the code. Thanks!