f4exb/sdrangel #192: Phosphor spectrum not working

antoine-roux, 2018-06-29:
Hello,

I'm unable to get the phosphor spectrum display working :(
When I enabled it, it just show a dark blue background but nothing more...
In the same time, the "max hold" feature is only tracing a line at -100 dB. I suppose it's linked.

[image]

I don't have anything relevant in the logs (log level = debug)
```
2018-06-29 22:59:07.163 (I) SDRangel 4.0.1 Qt 5.11.1 64b x86_64 Arch Linux DSP Rx:16b Tx:16b PID 12817
2018-06-29 22:59:07.423 (W) qUncompress: Input data is corrupted
2018-06-29 22:59:07.423 (W) qUncompress: Input data is corrupted
2018-06-29 22:59:07.423 (W) qUncompress: Input data is corrupted
2018-06-29 22:59:07.424 (I) PluginManager::loadPluginsDir: loaded plugin libchanalyzer.so
2018-06-29 22:59:07.425 (I) PluginManager::loadPluginsDir: loaded plugin libdemodam.so
2018-06-29 22:59:07.425 (I) PluginManager::loadPluginsDir: loaded plugin libdemodatv.so
2018-06-29 22:59:07.426 (I) PluginManager::loadPluginsDir: loaded plugin libdemodbfm.so
2018-06-29 22:59:07.442 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddatv.so
2018-06-29 22:59:07.444 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddsd.so
2018-06-29 22:59:07.444 (I) PluginManager::loadPluginsDir: loaded plugin libdemodlora.so
2018-06-29 22:59:07.445 (I) PluginManager::loadPluginsDir: loaded plugin libdemodnfm.so
2018-06-29 22:59:07.445 (I) PluginManager::loadPluginsDir: loaded plugin libdemodssb.so
2018-06-29 22:59:07.446 (I) PluginManager::loadPluginsDir: loaded plugin libdemodudpsrc.so
2018-06-29 22:59:07.446 (I) PluginManager::loadPluginsDir: loaded plugin libdemodwfm.so
2018-06-29 22:59:07.447 (I) PluginManager::loadPluginsDir: loaded plugin libmodam.so
2018-06-29 22:59:07.483 (I) PluginManager::loadPluginsDir: loaded plugin libmodatv.so
2018-06-29 22:59:07.484 (I) PluginManager::loadPluginsDir: loaded plugin libmodnfm.so
2018-06-29 22:59:07.485 (I) PluginManager::loadPluginsDir: loaded plugin libmodssb.so
2018-06-29 22:59:07.486 (I) PluginManager::loadPluginsDir: loaded plugin libmodudpsink.so
2018-06-29 22:59:07.486 (I) PluginManager::loadPluginsDir: loaded plugin libmodwfm.so
2018-06-29 22:59:07.487 (I) PluginManager::loadPluginsDir: loaded plugin liboutputbladerf.so
2018-06-29 22:59:07.488 (I) PluginManager::loadPluginsDir: loaded plugin liboutputfilesink.so
2018-06-29 22:59:07.489 (I) PluginManager::loadPluginsDir: loaded plugin liboutputhackrf.so
2018-06-29 22:59:07.490 (I) PluginManager::loadPluginsDir: loaded plugin liboutputsdrdaemonsink.so
2018-06-29 22:59:07.491 (I) PluginManager::loadPluginsDir: loaded plugin libinputairspy.so
2018-06-29 22:59:07.491 (I) PluginManager::loadPluginsDir: loaded plugin libinputbladerf.so
2018-06-29 22:59:07.492 (I) PluginManager::loadPluginsDir: loaded plugin libinputfcdpro.so
2018-06-29 22:59:07.493 (I) PluginManager::loadPluginsDir: loaded plugin libinputfcdproplus.so
2018-06-29 22:59:07.494 (I) PluginManager::loadPluginsDir: loaded plugin libinputfilesource.so
2018-06-29 22:59:07.494 (I) PluginManager::loadPluginsDir: loaded plugin libinputhackrf.so
2018-06-29 22:59:07.495 (I) PluginManager::loadPluginsDir: loaded plugin libinputrtlsdr.so
2018-06-29 22:59:07.496 (I) PluginManager::loadPluginsDir: loaded plugin libinputsdrdaemonsource.so
2018-06-29 22:59:07.496 (I) PluginManager::loadPluginsDir: loaded plugin libinputtestsource.so
2018-06-29 22:59:08.232 (I) WebAPIServer::start: starting web API server at http://127.0.0.1:8091
gf256_ctx::gf256_init_: initialized
```

Thanks again for this good piece of code :) !
Antoine

f4exb, 2018-07-02:
There is another issue: the frequency scale is centered on 0 and this could indicate some problem with the communication with the spectrum widget. I cannot reproduce myself but I am using an older version of Qt (5.9.5) so this could be an issue with the newer versions of Qt.

The +/-24 kHz range seems to indicate that the sample rate is received correctly. However this could be a default value. If you try to change the decimation factor what happens to the frequency scale?

antoine-roux, 2018-07-09:
Hello,

I don't know if it can helps, but when I'm compiling sdrangel with CMake 3.11.4, I got a warning about OpenGL saying it uses the legacy library:
```
CMake Warning (dev) at /usr/share/cmake-3.11/Modules/FindOpenGL.cmake:270 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
```
```
$ cmake --version
cmake version 3.11.4
```

If I'm not wrong, the phosphor spectrum uses OpenGL. Maybe it can helps...

See you,
Antoine

f4exb, 2018-08-05:
OpenGL is used for all graphical displays (spectrum, scope, TV screen, DV scope). You wouldn't be able to display any spectrum if OpenGL was broken.

f1ixx, 2018-08-26:
Hello
I can't display "phosphor" function et "max hold".
"phosphor" display back color dark blue
"max hold" only red line on down of spectrum.

I' don't try to debug, but i have found where is my problem. 
in file : sdrgui/gui/glSpectrum.cpp
in function : void GLSpectrum::updateHistogram(const std::vector<Real>& spectrum)
I'm comment block from 418 line to 482 line = #ifdef USE_SSE2 to #else
and the 495 line => #endif

Sorry for English
Boris

f4exb, 2018-08-28:
I am not sure that having explicit SSE2 code is still useful with modern compilers so I might get rid of it. The quick tests I have made show no performance difference.

It seems that in your case SSE2 is detected but the SSE2 code is not working for some reason. Leaving it to the compiler would circumvent the issue.

f1ixx, 2018-08-28:
Hello
I'm try to understand the SSE function and why it don't work.
After some printf, some test and some external test. I'm found this (works for me)

in file /sdrgui/gui/glspectrum.cpp :
```
const __m128 refl = {m_referenceLevel, m_referenceLevel, m_referenceLevel, m_referenceLevel};
const __m128 power = {m_powerRange, m_powerRange, m_powerRange, m_powerRange};
const __m128 mul = {100.0f, 100.0f, 100.0f, 100.0f};
```
i'm remove `const` and change to :
```
__m128 refl = {m_referenceLevel, m_referenceLevel, m_referenceLevel, m_referenceLevel};
__m128 power = {m_powerRange, m_powerRange, m_powerRange, m_powerRange};
__m128 mul = {100.0f, 100.0f, 100.0f, 100.0f};
```
and it's ok. Why? I'm not coder for know.
Performance, i can't see difference on CPU use. Test with different configuration on SDR dongle.

Mais un enorme merci  pour votre logiciel et votre enorme travaille pour fournir tout ca.

Boris, F1IXX

Fedora 28
gcc-c++-8.1.1-5.fc28.x86_64
qt5-qtbase-gui-5.10.1-6.fc28.x86_64

f4exb, 2018-08-29:
Hello Boris,
thanks for your investigation. I also noticed hardly any difference between SSE code and plain C++ code. I think the compiler does optimization with vector code by itself and often does it even better. If AVX or AVX2 is available it might use 256 bit instructions. So I'm still wondering if coding vector code is really useful.
Best regards, Edouard.

f4exb, 2018-10-06:
This SSE2 code will be removed in next version.