f4exb/sdrangel #193: windows builds/downloads are behind linux ones

beta-tester, 2018-07-04:
hi,
on [f4exb/sdrangel - Latest release](https://github.com/f4exb/sdrangel/releases),
there is no windows release available for sdrangel build 4.0.1 and 4.0.2.
does it mean, that windows support is stopped or broken or do those builds contains only fixes for linux?

beta-tester, 2018-07-04:
oh, forgett it - just saw that your dev environment smoked.
sorry to hear.

closed 2018-07-04 by beta-tester

f4exb, 2018-07-05:
Still facing issues with the hard disk where the Windows VM used to live. Waiting for parts.

beta-tester, 2018-07-05:
oh man... you have all the work and now all the pain...
and i have nothing better to do as to cry.

thank you for your work.

PS.: maybe i should better do a backup of my hard disk... before it will smoke as well.

beta-tester, 2018-07-08:
it brings tears to my eyes...
but i respect your decision.

beta-tester, 2018-07-08:
now you can change it from "maybe" to "is"
[image]

f4exb, 2018-07-11:
I eventually managed to have a working environment for the build so as you may notice you now have an archive for v4.0.3:
https://github.com/f4exb/sdrangel/releases/tag/v4.0.3