f4exb/sdrangel #195: P25 demod support

sparklespdx, 2018-07-10:
I'm mostly creating this issue to gauge interest in a P25 plugin for sdrangel. This is certainly something that has been done before, more recently: http://osmocom.org/projects/op25/wiki.

I'm new to signal processing, I think that listening to P25 is pretty cool and I am interested in seeing what it would take to create this plugin.

f4exb, 2018-07-10:
The main point is to make it a feature of dsdcc: https://github.com/f4exb/dsdcc
Then there is a little work to integrate the status messages specific to P25 in DSD demodulator.

sparklespdx, 2018-07-10:
Totally makes sense. Seems like I should add an issue to that repo? We can leave this open for visibility or close, I'm good with either one.

f4exb, 2018-07-11:
Yes please. In fact I did not open it myself since I have other priorities. Consequently I cannot promise a quick implementation. It just happens that MMDVM has P25 support so provided I can switch on a P25 reflector and it has some traffic I would be able to get sample frames. I recently did the same thing for NXDN.

sparklespdx, 2018-07-12:
I actually would like to get up to speed and learn how to conbtribute this
myself! I'll make an issue in the other repo.

gaspar1987, 2019-07-05:
P25 in DSD demodulator mettre sur le  4.10.5

gaspar1987, 2019-07-09:
bonjour monsieur  il faux mettre de la modulation P25 EN LD T 1 LD 2  dans le dsd démodulateur pour la trame du p25 ACOP compatible avec motorola apx xtl saber etc ..

f4exb, 2022-02-10:
dropped