f4exb/sdrangel #198: PLUTO transmit power level

W3IP, 2018-07-14:
I have been testing my PLUTO transmitter with SDRAngel version 4.0.0 and 4.0.3 and find the power output using any modulator is consistently 24 db (4 bits) low (i.e. -18 dBm out instead of + 6 dBm out). The tx attenuator is set to 0 dB. Looking through the source code, I see the line of code that appears to left justify the 12 bit symbol in the 16 bit signed integer has been commented out (plutosdroutputthread.cpp line 97). Would it be possible to enable this code to achieve full transmitter output?

Thanks, Mike

f4exb, 2018-07-16:
Did you measure the power output? 

A conversion is done at line 84 with the convert method and it actually does right the opposite because what is expected is that the 16 bit wide symbols built in the application are converted into the supposedly 12 bit symbols of the Pluto. Yet the txChannelConvert makes a call to a libiio function and maybe something happens there. In any case it is not as simple as uncommenting line 97.

f4exb, 2018-07-16:
Line 57 in plutosdroutputthread.h sets the conversion scheme so to stay with 16 bits one would have to code instead (note the "16"): 
`Interpolators<qint16, SDR_TX_SAMP_SZ, 16> m_interpolators;`

f4exb, 2018-07-16:
It seems this is the issue. Some conversion takes place in iio_channel_convert_inverse and it must also convert from the 16 bit samples of the host to the 12 bit samples of the Pluto so one should not do it on the host side. Changing the code in plutosdroutputthread.h as mentioned above seems to fix the problem. Although I didn make precise masurements it does seem to have much more power for the same attenuator setting

W3IP, 2018-07-18:
The PLUTO transmitter now checks out correctly at +6 to +7 dBm output!

Thanks again - 73 Mike, W3IP

closed 2018-07-18 by W3IP