f4exb/sdrangel #203: entry point error

TonyHagen, 2018-07-20:
When i start the transmitter a pop up window appears " Entry point error, not able to locate some .dll files"

What could be the reason ?
-Tony

f4exb, 2018-07-20:
Without any more information I conclude at a bad installation.

closed 2018-07-20 by f4exb

TonyHagen, 2018-07-21:
check here too the first post – the same has been reported https://www.rtl-sdr.com/a-tutorial-on-using-sdrangel-for-dmr-d-star-and-fusion-reception-with-an-rtl-sdr/

Fuurther what I understand you should have 64bit os and system and the sdrangel software folder needs to be placed in the program folder and run the sdrangel application – SO WHERE THE INSTALLATION IS INVOLVED

OTHERWISE my hackrf is working perfect

73
Tony

From: f4exb 
Sent: Saturday, July 21, 2018 4:20 AM
To: f4exb/sdrangel 
Cc: TonyHagen ; Author 
Subject: Re: [f4exb/sdrangel] entry point error (#203)

Without any more information I conclude at a bad installation.

TonyHagen, 2018-07-21:
IS it compatible with hackrf

73
Tony

From: f4exb 
Sent: Saturday, July 21, 2018 4:20 AM
To: f4exb/sdrangel 
Cc: TonyHagen ; Author 
Subject: Re: [f4exb/sdrangel] entry point error (#203)

Without any more information I conclude at a bad installation.

f4exb, 2018-07-21:
If you have such questions particularly regarding running on Windows please post a message to https://groups.io/g/sdrangel to seek help from other users.