f4exb/sdrangel #204: PLUTO transmitter anomalous behavior

W3IP, 2018-08-04:
When using the SSB modulator on PLUTO (version 4.0.4): On startup, the transmitter initializes correctly with no power output. Click on the tone modulation button, the tone comes on at the correct power level and correct frequency offset. Click the tone modulation  button again. Instead of returning the transmitter to no power out, the tone moves to 0 Hz offset and the power level remains the same. 

Now click on the Morse keyer button with no text in the text box. The transmitter returns to no power out. Type some text, click the send text and the play text in a loop buttons, and the CW keyer works properly at the requested offset and power output.  Click the Morse keyer button again to stop the CW output, and the output will either go to no power output if you stopped in keyer while it was sending a space, or to 0 Hz offset at the requested power output level if you stopped the keyer while it was sending a dot or dash. At the moment, the only way to guarantee the PLUTO transmitter actually returns to no power out state is to turn on and off the audio input (mic) button with no audio input source connected.

f4exb, 2018-08-05:
Changed the title: the issue is more likely to be with the modulator plugin rather than with the type of Tx used. It seems that when there is no input the DC remains which causes a carrier to appear at channel frequency 0. This is cancelled with the "mute" button.

f4exb, 2018-08-11:
Implemented in v4.0.6

closed 2018-08-11 by f4exb