f4exb/sdrangel #205: recording broken

beta-tester, 2018-08-17:
in version 4.0.6 seems the recording is broken.
i just recorded a huge file with sdrangel 4.0.6 and tried to play it back,
but the shown frequency "1,000,000kHz",
the shown bandwidth "0.058k",
the shown time "1970-03-15"
and the shown length are totally wrong
and nothing is shown on the waterfall nor the spectrum view.

i tried an older similar huge recoding made with 4.0.4 and there i can playback the recording correctly on 4.0.6.

damn stupid me, i updated sdrangel without testing if the recording will succeed, just before i recorrded an importand one-time-ever-event... (installing and setting up the wireless consumption meter of my  radiator)
now i guess i lost everything i wanted to record/playback/analyse... :((

i used the windows version 4.0.6 of sdrangel.
with my PlutoSDR.

[image]

beta-tester, 2018-08-17:
maybe it is depending only on the file-size.
the faulty recording is 17GB big, the file with 12GB is working.

f4exb, 2018-08-18:
Well I have not encountered this issue just by changing the record name with a 'ö' in the name. So it must be something else that corrupted the header of the file. The first few bytes contain time and sample rate information that was obviously damaged.

beta-tester, 2018-08-19:
the file name that wasn't read correctly was moved and renamed fust after recording and before i tried to load in FileSource: 
from `E:\SDR\sdrangel\rec0_2018-08-17T05_51_40_895.sdriq`
to `E:\SDR\rec0_2018-08-17T05_51_40_895-Heizkörper08-13.sdriq`

the working dir and path of sdrangel is:
`E:\SDR\sdrangel\`
`E:\SDR\sdrangel\sdrangel.exe`

filesystem is: `ntfs`

every think went good with renaming the file to:
`E:\SDR\rec0_2018-08-17T05_51_40_895-Heizkoerper08-13.sdriq`

so the file header/content is not damaged.

i think the FileSource wasn't reading the file header at all, because i get similar values shown in the screenshot, when i try to read a file with 0Byte into the FileSource.

e.g.
`E:\SDR\null.sdriq` with 0byte shows 00:00:00 [Total record time]
[image]

`E:\SDR\null-ö.sdriq` with 0byte shows 01:17:38 [Total record time]
[image]

`E:\SDR\null-ö-2.sdriq` with 0byte shows 10:08:57 [Total record time]
[image]

different values just by changing the name of the file only.

f4exb, 2018-10-08:
To be sure the file header needs a CRC to protect it in case of corruption.
Closed in favor of #206

closed 2018-10-08 by f4exb

gaspar1987, 2019-06-30:
bonsoir  le record  qui marche pas  parque que il enregistrent pas .

PR #351 (2019-05-27): Use correct encoding for file names in FileSource plugin
Opening of file named `Ы.sdriq` with FileSource plugin on Windows platform results in error.
To fix this either `std::wstring` should be used in `m_ifstream.open()` call or code should be rewritten to use `QFile` instead of `std::ifstream`.
This PR implements `wstring` solution. Hope it will not break other platforms' build.

Fixes #205.
commit cf89e20f (2019-05-27): Use correct encoding for file names in FileSource plugin (fixes #205)