f4exb/sdrangel #206: FileSource: play time and slider broken/overflow

beta-tester, 2018-08-17:
hi, its me FileSource, when i play a recorded file, somewhere after 10 minutes the playtime [Record time from start] and the slider where you can quickly jump to a specific time are broken.
(the slider jumps to beginning and the time is shown like an overflow happend)
[image]

f4exb, 2018-10-08:
I am not able to reproduce the problem but analyzing the code reveals quite a few vulnerabilities.

1. There is no CRC for the header but the correct functioning relies only on this header so if it gets corrupted the thing goes havoc
2. The starting timestamp is recorded as a size_t but the exact size of this type is either 4 or 8 bytes depending on the target for which the code is compiled. On most Linux systems it will be 8 (64 bits) but not on armv7. On Windows since the toolchain is 32 bits (MinGW32) this will be 4. This should be unified taking a guaranteeed 8 bytes type (uint64_t)
3. In filesourceinput.cpp at line 130 it is assumed that the sample size is 4 bytes but this is only true for 16 bits samples. If the record was made from a 24 bit version then the size is doubled which results at best in wrong readings and possibly crashing because of trying to read past the end.

The use of CRC implies the creation of a conversion script or program to recycle older records.

f4exb, 2018-10-09:
Another point:
  4. There is no pragma pack around the structure so it may get padded randomly depending on the compiler

beta-tester, 2018-10-10:
hi, i tested the version 4.2.0 under windows 10 x64. same behavior as before.

i don't want to upload a test sdriq file, because of its huge size.
but you can create your own test file...
[image]

1. open sdrangel
2. choose TestSource as source
3. change DEC to 1, Sz to 12bits, SRR to 2,500,000S7s
4. start the source
5. start recording
6. after 35 minutes stop recording

now you have a test file of ~19.5GByte

with that file you should be able to reproduce that behavior (at least on windows).

f4exb, 2018-10-15:
Fix implemented in v4.2.2

closed 2018-10-15 by f4exb