f4exb/sdrangel #207: [Display max hold] with infinite holfoff?

beta-tester, 2018-08-17:
hi,
is there an option existing to have an **infinite** holfoff for the red [Display max hold] line in the spectrum view?
[image]

sometimes there is traffic in the spectrum very very rarely. with an infinite holdoff of the [Display max hold] it would be possible to let the spectrum running for a day. and comming back later to watch where the red line was pulle up. everywhere, where the [Display max hold] line was pulled up, there was traffic.
then i can optimize my listening process to those specific frequencies.
or at least, then i know, if there is traffic to expect or not.

f4exb, 2018-08-19:
If this is of any use to you you can also use the spectrum fixed average with the largest possible averaging (1000 samples). Depending on the base band sample rate it can record a long time on the waterfall. If you display only the waterfall on a 1920x1080 display you have ~960 lines. For example with a sample rate of 96 kS/s a FFT of 4k (4096) and an averaging of 1k (1000) there is one FFT line for each: (1/96000)*4096*1000 = 42.7s with 960 lines this makes a 42.7*960 = 40960s time span which is 11h22m40s
Below screenshot shows a portion of the 20m band. It was taken with a FFT of 1k (1024) and a shorter waterfall due to the presence of the spectrum line. There is still over 2h display:
[image]
The number of samples for the fixed averaging could be extended if necessary.

beta-tester, 2018-08-19:
thank you, you suggestion,
but the traffic i try to observe are data packets of some few milli second. 
they are not only very very rarely, they are also very short.
it looks like averaging will not leave a print on the waterfall for that short and tiny data packets.
only longer data packets or multiple data packets in a shorter time will be shown on the waterfall.
so i possibly miss some events of traffic... with a high averaging value.

beta-tester, 2018-08-20:
at zero averaging i can see the small data packets in the red line and in the waterfall.
that means they are long enough to get detected for the graphics.
but as soon i increase the averaging more an more, the less is the effect of leaving a visible print in the graph until nothing of the data packet is visible.

but maybe you are right and i missed already even shorter data packets.

f4exb, 2018-08-21:
I think this is an interesting use case but to be handled properly the "max hold" should be implemented in the pre-processing like the averaging. I will look into this.

f4exb, 2018-10-15:
Implemented in v4.2.2: added a "Max" mode to the possible averaging modes. Although not doing average actually it works similarly to the "Fix" average but taking the maximum over the averaging range.

beta-tester, 2018-10-15:
sdrangel 4.2.2 (Windows) crashes or gets unresponsive at [Number of avaraging samples] above ~10k.
doesn't matter what avaraging type is selected [No, Mov, Fix, Max]

tested under Windows 10 x64 with 16GByte RAM,
RTL-SDR dongle @ 8bit, 2400000S/s,
PlutoSDR @ 12bit, 2500000S/s.

beta-tester, 2018-10-15:
before you spend too much time to that Max-implementation of 4.2.2...
as far as i could see, it doesn't look like to give me the ability to see traffic of tiny pulses on long term observations.

or i use that functionality in a wrong way.

what i am looking for is something like the red [Display max hold] line, that already exists,
but with the ability to turn up the  holdoff time to infinite (by the existing holdoff "turn-knop" or an additional control like a check box or something similar to enable/disable infinite holdoff time).

f4exb, 2018-10-16:
10000 samples (FFT that is) can be a very large number. The tooltip over the number tells you the actual time before you get a new value. Ex: with a FFT of 1k (1024) and 10000 samples at 2.5 MS/s this is 1024*10000 / 2500000 = 4.096 s. This supposes that there is no decimation. If there is any you will have to multiply this number by the decimation factor. In any case make sure you wait that time before expecting to see anything on the spectrum. 

Also Windows is a bad platform anyway with a lot of bugs maybe due to the toolchain used by Qt maybe due to the O/S itself. Please switch to Linux for any not so casual usage. I generally do not give support for Windows related bugs.

I have no issues with a 10k accumulation value at 1k FFT and 3 MS/s without decimation. Linux Ubuntu 18.04 also 16 GB RAM. "avg time" is 3.41s and it already looks very very slow.

beta-tester, 2018-10-16:
hi,
made a pull request with an additional button to prevent the red hold line from dropping down permanently.
so i can see if there was traffic in the spectrum after a long term observation,
without using the averaging type switches.

BTW: even under ubuntu 18.04 LTS 64bit, 16GB RAM sdrangel crash,
when i turn up the [Number of avaraging samples] to its maximum value of 1M.
```
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
```

f4exb, 2018-10-16:
I still have to think of the max hold for the red line of the spectrum display (that's what it is actually). Moreover "holdoff" is an improper term so the whole thing has to be reworked.

beta-tester, 2018-10-16:
with the functionality of that pull request #228, i am pretty happy and that was all i wanted...

i am sorry for guiding you in a totally wrong direction with the term Max Hold...
i am not that good in explaining of what i am looking for.
hopefully you did not waste too much work and time in the new averaging functionality.

when i made settings that makes no sense, that's because i don't know and care the math behind...
(shame on me).
when there is an option, i try out mostly its limits and see what's happen...

f4exb, 2018-10-17:
"Max hold" is the thing you are looking for actually and what should be implemented. I have to look back at the original code for what was expected by "holdoff" which control has been added by me later without understanding what it actually meant. Probably this control is not needed and the real estate can be recovered for a max hold button. It is still an issue if the bursts are shorter than the refresh period and the "Max" over a range of FFT is very useful. I am using it successfully to monitor DME bursts for example.

f4exb, 2018-10-17:
That's it a decay of 0 if enabled to do what it should (code needs a change here)  just does the same thing by keeping the maximum forever. 

After a second look the "holdoff" seems up to its meaning by excluding from event history the too transient changes. This removes too transient contributions.

f4exb, 2018-10-18:
It turns out that this "holdoff" thing makes simply a decay divisor. The whole spectrum phosphor display controls need to be reworked in a more straightforward way.

f4exb, 2018-10-20:
Implemented a decay of 0 in v4.2.3 this effectively keeps the max hold until the clear histogram button is pressed.

closed 2018-10-20 by f4exb

beta-tester, 2018-10-21:
it works...
nice... thank you... great job! 👍  🥇