f4exb/sdrangel #208: Pulse-audio issue?

angrysmiley, 2018-08-20:
When selecting any pulseaudio device I'm getting the following error: 

2018-08-20 17:46:39.744 (I) AudioDeviceManager::setOutputDeviceInfo: unknown device alsa_output.pci-0000_00_14.2.analog-stereo
2018-08-20 17:46:39.744 (W) AudioDeviceManager::setOutputDeviceInfo: index: 58 device: alsa_output.pci-0000_00_14.2.analog-stereo no FIFO registered yet hence no audio output has been allocated yet

There is no sound whatsoever in sdrangel (other programs/software using pulseaudio work fine) 

Setup is as follows:    SDRangel 4.0.6 Qt 5.11.1 64b x86_64 Debian GNU/Linux buster/sid

f4exb, 2018-09-13:
index: 58 means it tries to access the 59th device in the list of available devices. This is a very large number and I would first inspect my audio setup to see if there are actually more than 59 devices. I have never been able to reproduce such a case.

f4exb, 2018-09-13:
You should also try to activate the debug level in the console (Preference -> Logging) to get more information.

closed 2018-10-06 by f4exb