f4exb/sdrangel #212: USB2.0 Tx issues

KC7NOA, 2018-08-31:
I had a separate open question about the antenna selection freezing a GUI but everything is working now when getting a USB 3.0.

In that thread I comment that I can run the sample rate down to 1.5 Ms/s  but using the same settings on USB 2.0 will not work for Tx.

f4exb, 2018-08-31:
What's the point of this ticket? Do you expect some code fix? I suppose you're dealing with LimeSDR. It is well known that LimeSDR is using USB 3.0.

KC7NOA, 2018-09-13:
Are you trying to tell me that as long as you have USB3.0 you can tx with lower through put than USB2.0... but no Tx support with USB2.0 set to the same capture rates?

f4exb, 2018-09-13:
USB 2.0 or 3.0 is irrelevant. USB interface is handled by libusb and the lower level drivers. I still don't see the point of this ticket.

KC7NOA, 2018-09-15:
Because Tx does not happen on USB2.0 and USB3.0 does ... it looks like the program freezes up when Tx is started on USB2.0 but works on USB2.0.

KC7NOA, 2018-09-16:
Perhaps this is the same problem seeing that you describe something about mutex in another bug thread

KC7NOA, 2018-09-18:
Just as a side note

Same hardware with added D720202 mPCIe to USB3.0 does manage Tx with no hangup or freezing

KC7NOA, 2018-10-06:
Nonissue but does not work ... ok

f4exb, 2018-10-07:
This is an issue with your setup I have no issues running any Tx on USB2. This is not in my software hence I have nothing to do about it.