f4exb/sdrangel #214: SSB audio compressor: default settings generate feedback

RipperFox, 2018-09-06:
Tested v4.0.7 on Windows with Pluto sink - CMP seems almost unusable for me.

[image]

See video here: https://youtu.be/xxj7Rzq0h2I (Audio was recorded from RF via another SDR.)

f4exb, 2018-09-06:
Not sure it is entirely related but you have a very very large sample rate in the baseband for just SSB judging by the number at the top left corner of the screenshot and confirmed by the frequency scale. 2500 kS/s is a lot when you just need 48 kS/s. This puts a lot of strain on the channel interpolator that is not designed to match such a large gap. You should try to increase software decimation on the Pluto device interface.

RipperFox, 2018-09-06:
Just tried with these settings - same distortion:

Same signal 
[image]

RipperFox, 2018-09-07:
I tested a but further: Problem exists with FileSync & Source, too.
It seems there is some gain routine going wild? Even when there's no input from the mic (I tested muting the device in Windows and by selecting a unconnected virtual audio cable) there's a build up of noise..

g0hww, 2019-02-22:
I'm not entirely certain how the compressor should be set up, but i have had similar issues with instability. I could do with a bit more oomph in my HackRF-based uplink for Es'hail-2, which has netted me a QSO using sdrangel for SSB tx, with a 13dB "4W" wifi amp and a 21dBi 50cm dish :)

f4exb, 2019-04-28:
Cannot reproduce.

closed 2019-04-28 by f4exb