f4exb/sdrangel #217: Build error

cyberxml, 2018-09-20:
Building on Fedora 28 with current git pull of sdrangle and bladeRF from github
Linux localhost.localdomain 4.17.14-202.fc28.x86_64 #1 SMP Wed Aug 15 12:29:25 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

https://www.github.com/nuand/bladeRF

Make error

> [ 58%] Building CXX object plugins/samplesource/bladerfinput/CMakeFiles/inputbladerf.dir/bladerfinput.cpp.o
> In file included from /home/cyberxml/gits/sdrangel/plugins/samplesource/bladerfinput/bladerfinput.h:23,
>                  from /home/cyberxml/gits/sdrangel/plugins/samplesource/bladerfinput/bladerfinput.cpp:17:
> /home/cyberxml/gits/sdrangel/plugins/samplesource/bladerfinput/bladerfinput.cpp: In member function ‘bool BladerfInput::openDevice()’:
> /home/cyberxml/gits/sdrangel/plugins/samplesource/bladerfinput/bladerfinput.cpp:115:43: error: invalid conversion from ‘bladerf_channel’ {aka ‘int’} to ‘bladerf_channel_layout’ [-fpermissive]
>      if ((res = bladerf_sync_config(m_dev, BLADERF_MODULE_RX, BLADERF_FORMAT_SC16_Q11, 64, 8192, 32, 10000)) < 0)
>                                            ^~~~~~~~~~~~~~~~~
> /usr/local/include/libbladeRF.h:2485:58: note:   initializing argument 2 of ‘int bladerf_sync_config(bladerf*, bladerf_channel_layout, bladerf_format, unsigned int, unsigned int, unsigned int, unsigned int)’
>                                    bladerf_channel_layout layout,
>                                    ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
> make[2]: *** [plugins/samplesource/bladerfinput/CMakeFiles/inputbladerf.dir/build.make:80: plugins/samplesource/bladerfinput/CMakeFiles/inputbladerf.dir/bladerfinput.cpp.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:3587: plugins/samplesource/bladerfinput/CMakeFiles/inputbladerf.dir/all] Error 2
> make: *** [Makefile:130: all] Error 2

cmake ..

> CMake Deprecation Warning at CMakeLists.txt:2 (cmake_policy):
>   The OLD behavior for policy CMP0043 will be removed from a future version
>   of CMake.
> 
>   The cmake-policies(7) manual explains that the OLD behaviors of all
>   policies are deprecated and that a policy should be set to OLD only under
>   specific short-term circumstances.  Projects should be ported to the NEW
>   behavior and not rely on setting a policy to OLD.
> 
> 
> -- The C compiler identification is GNU 8.1.1
> -- The CXX compiler identification is GNU 8.1.1
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> CMake Warning (dev) at /usr/share/cmake/Modules/FindOpenGL.cmake:270 (message):
>   Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
>   available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
>   cmake_policy command to set the policy and suppress this warning.
> 
>   FindOpenGL found both a legacy GL library:
> 
>     OPENGL_gl_LIBRARY: /usr/lib64/libGL.so
> 
>   and GLVND libraries for OpenGL and GLX:
> 
>     OPENGL_opengl_LIBRARY: /usr/lib64/libOpenGL.so
>     OPENGL_glx_LIBRARY: /usr/lib64/libGLX.so
> 
>   OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
>   compatibility with CMake 3.10 and below the legacy GL library will be used.
> Call Stack (most recent call first):
>   CMakeLists.txt:60 (find_package)
> This warning is for project developers.  Use -Wno-dev to suppress it.
> 
> -- Found OpenGL: /usr/lib64/libOpenGL.so   
> -- Found PkgConfig: /usr/bin/pkg-config (found version "1.4.2") 
> -- Boost version: 1.66.0
[... 109 lines cut ...]
>   of the following names:
> 
>     OpenCVConfig.cmake
>     opencv-config.cmake
> 
>   Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
>   "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
>   provides a separate development package or SDK, be sure it has been
>   installed.
> 
> 
> -- Found Lime Suite: /usr/local/include, /usr/local/lib/libLimeSuite.so
> -- Checking for module 'libcm256cc'
> --   Package 'libcm256cc', required by 'virtual:world', not found
> -- Found CM256cc: /usr/local/include/cm256cc, /usr/local/lib64/libcm256cc.so
> -- SDRdaemonFEC: use SSSE3 SIMD
> -- Found Lime Suite: /usr/local/include, /usr/local/lib/libLimeSuite.so
> -- Checking for module 'libcm256cc'
> --   Package 'libcm256cc', required by 'virtual:world', not found
> -- Found CM256cc: /usr/local/include/cm256cc, /usr/local/lib64/libcm256cc.so
> -- SDRdaemonSource: use SSSE3 SIMD
> -- LibMiriSDR found
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/cyberxml/gits/sdrangel/build

cyberxml, 2018-09-20:
Is this related to 'v2' of libbladerf?
https://github.com/f4exb/sdrangel/issues/215

cyberxml, 2018-09-20:
Work around was to install "bladeRF-2016.06" into a local directory.
Note that I had to modify two files to compile on my Fedora 28 distro
1. device_identifier.c
2. flash.c

The run cmake with pointers to the older bladerf library/include 
cmake -DLIBBLADERF_LIBRARIES=/home/cyberxml/local/bladeRF-2016.06/usr/local/lib64/libbladeRF.so -DLIBBLADERF_INCLUDE_DIR=/home/cyberxml/local/bladeRF-2016.06/usr/local/include ..

I have not yet tested this with a bladeRF as I was compiling to get support for another device.

f4exb, 2018-09-20:
Support for libbladerf v2 is under way with support of the BladeRF micro. You can try the dev branch which should be working with libbladerf 2018.08 and the version 1 of the BladeRF (plugins are renamed with bladerf1 in theit name).

f4exb, 2018-10-06:
Closed in favor of #215

closed 2018-10-06 by f4exb