f4exb/sdrangel #218: RESTful APIs

Mitchell1314, 2018-09-22:
Hi sir,
    The restful APIs you developed highly arose my great interest. Through invoking these APIs, I can easily interact with sdrangel application, such as start or stop a device.  But more functions are in my interest. For example, Can I obtain spectrum data or audio data by invoking RESTful APIs? In summary, can I aquire information about spectrum analysis and signal demodulation by invoking your RESTful APIs? Waiting for your reply. Thank you.

f4exb, 2018-09-24:
So to summarize I do not intend to make spectrum nor audio data available through REST API. This kind of data is better handled via streaming which is already the case for audio. Spectrum may follow in the future. The exact form of the spectrum stream is not defined yet.

f4exb, 2018-10-05:
Closed in favor of #224

closed 2018-10-05 by f4exb