f4exb/sdrangel #219: Add Lime SDR Mini support for windows?

wgaylord, 2018-09-25:
I see that Lime SDR Mini support is disabled for windows. Any chance that this will ever be fixed?

Or do I have to manually compile SDRAngle to be able to use it on windows with a Lime SDR Mini?

f4exb, 2018-09-26:
The doc says "LimeSDR-Mini seems to have problems with Zadig driver therefore it is supported in Linux only."

I do not intend to support it via other means that libusb and therefore Zadig so basically this is out of my control. Even if you would compile it yourself you would hit the brick wall with Zadig.

closed 2018-09-26 by f4exb

wgaylord, 2018-09-26:
I am confused what Zadig has to do with this? As my LimeSDR Mini used the FTDI driver that windows provided.  Also LimeSuite can see and use my LimeSDR Mini just fine.

I didn't have to use Zadig at all.

f4exb, 2018-09-26:
With libusb and MinGW toolchain you have to use Zadig. Stock LimeSuite is compiled with MSVC libraries while SDRangel toolchain for Windows is based on MinGW. Binaries produced by one is not compatible with the other. There are no plans to change this.

wgaylord, 2018-09-26:
I guess it’s time to get dual booting working. As I tried to use a virtual machine to run it but it won’t attach the LineSDR Mini (another common problem.)

Or move to just use one of the ones packaged with Pothos

wgaylord, 2019-05-24:
Is this still a won't fix?

f4exb, 2019-05-24:
Yes