f4exb/sdrangel #220: XTRX not working

hmne, 2018-09-26:
**sdrangel**
26.09.2018 03:09:27.233 (W) QMetaObject::connectSlotsByName: No matching signal for on_sampleSource_changed()
26.09.2018 03:09:27.233 (W) QMetaObject::connectSlotsByName: No matching signal for on_sampleSink_changed()
26.09.2018 03:09:27.233 (W) QMetaObject::connectSlotsByName: No matching signal for on_channel_addClicked(bool)
26.09.2018 03:09:27.233 (W) QMetaObject::connectSlotsByName: No matching signal for on_action_reloadDevices_triggered()
26.09.2018 03:09:27.236 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libdsdcc.so'
26.09.2018 03:09:27.236 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libfcdhid.so'
26.09.2018 03:09:27.236 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libfcdlib.so'
26.09.2018 03:09:27.236 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libhttpserver.so'
26.09.2018 03:09:27.236 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libLimeSuite.so.18.06.1'
26.09.2018 03:09:27.236 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/liblogging.so'
26.09.2018 03:09:27.236 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libmirsdrapi-rsp.so.2.13'
26.09.2018 03:09:27.236 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libsdrbase.so'
26.09.2018 03:09:27.236 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libsdrgui.so'
26.09.2018 03:09:27.237 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libswagger.so'
26.09.2018 03:09:27.237 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libusb3380.so.0.0.1'
26.09.2018 03:09:27.237 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libxtrx.so.0.0.1'
26.09.2018 03:09:27.237 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libxtrx_lms7002m.so.0.0.1'
26.09.2018 03:09:27.237 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libxtrxdevice.so'
26.09.2018 03:09:27.237 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libxtrxdsp.so.0.0.1'
26.09.2018 03:09:27.237 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libxtrxll.so.0.0.1'
26.09.2018 03:09:27.237 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libxtrxll_libusb3380.so.0.0.1'
26.09.2018 03:09:27.237 (W) PluginManager::loadPlugins:  Failed to extract plugin meta data from '/usr/local/lib/libxtrxll_pcie.so.0.0.1'
26.09.2018 03:09:27.239 (I) PluginManager::loadPlugins: loaded plugin libchanalyzer.so
26.09.2018 03:09:27.240 (I) PluginManager::loadPlugins: loaded plugin libchanalyzerng.so
26.09.2018 03:09:27.240 (I) PluginManager::loadPlugins: loaded plugin libdemodam.so
26.09.2018 03:09:27.241 (I) PluginManager::loadPlugins: loaded plugin libdemodatv.so
26.09.2018 03:09:27.243 (I) PluginManager::loadPlugins: loaded plugin libdemodbfm.so
26.09.2018 03:09:27.244 (I) PluginManager::loadPlugins: loaded plugin libdemodlora.so
26.09.2018 03:09:27.245 (I) PluginManager::loadPlugins: loaded plugin libdemodnfm.so
26.09.2018 03:09:27.246 (I) PluginManager::loadPlugins: loaded plugin libdemodssb.so
26.09.2018 03:09:27.247 (I) PluginManager::loadPlugins: loaded plugin libdemodtcpsrc.so
26.09.2018 03:09:27.248 (I) PluginManager::loadPlugins: loaded plugin libdemodudpsrc.so
26.09.2018 03:09:27.249 (I) PluginManager::loadPlugins: loaded plugin libdemodwfm.so
26.09.2018 03:09:27.250 (I) PluginManager::loadPlugins: loaded plugin libmodam.so
26.09.2018 03:09:27.756 (I) PluginManager::loadPlugins: loaded plugin libmodatv.so
26.09.2018 03:09:27.757 (I) PluginManager::loadPlugins: loaded plugin libmodnfm.so
26.09.2018 03:09:27.758 (I) PluginManager::loadPlugins: loaded plugin libmodssb.so
26.09.2018 03:09:27.759 (I) PluginManager::loadPlugins: loaded plugin libmodudpsink.so
26.09.2018 03:09:27.759 (I) PluginManager::loadPlugins: loaded plugin libmodwfm.so
26.09.2018 03:09:27.760 (I) PluginManager::loadPlugins: loaded plugin liboutputfilesink.so
26.09.2018 03:09:27.761 (I) PluginManager::loadPlugins: loaded plugin libinputfcdpro.so
26.09.2018 03:09:27.762 (I) PluginManager::loadPlugins: loaded plugin libinputfcdproplus.so
26.09.2018 03:09:27.762 (I) PluginManager::loadPlugins: loaded plugin libinputfilesource.so
26.09.2018 03:09:27.763 (I) PluginManager::loadPlugins: loaded plugin libinputrtlsdr.so
26.09.2018 03:09:27.764 (I) PluginManager::loadPlugins: loaded plugin libinputxtrx.so
03:09:27.770385 INFO:    Probing 'libxtrxll_pcie.so.0' low-level library
03:09:27.770700 INFO:    Probing 'libxtrxll_libusb3380.so.0' low-level library
03:09:27.936130 INFO:    USB3380 dual fly GPEP RX mode is on, TX mode is on
03:09:27.936803 INFO:    USB3380: USB338x Speed: 480 Mbps
03:09:27.949861 INFO:    USB3380: Initialized PCIe dev 0100 [10ee:7012]
03:09:27.950071 INFO:    USB3380: IO thread started
03:09:27.950553 ERROR:   XTRX P: 00000066 00000000 <-> 00000000
03:09:27.950913 ERROR:   XTRX P: 00000066 00000000 <-> 00000000
03:09:27.950940 INFO:    XTRX USB3_1: RX DMA STOP MIMO (BLK:0 TS:0); TX DMA STOP MIMO
03:09:27.951114 INFO:    XTRX USB3_1: Device `/dev/xtrx0` was opened
03:09:28.064906 ERROR:   XTRX USB3_1: SPI[0/1] INT err:-110!
03:09:28.165656 ERROR:   XTRX USB3_1: SPI[0/1] INT err:-110!
03:09:28.266561 ERROR:   XTRX USB3_1: SPI[0/1] INT err:-110!
03:09:28.367416 ERROR:   XTRX USB3_1: SPI[0/1] INT err:-110!
03:09:28.468310 ERROR:   XTRX USB3_1: SPI[0/1] INT err:-110!
03:09:28.569151 ERROR:   XTRX USB3_1: SPI[0/1] INT err:-110!
03:09:28.670067 ERROR:   XTRX USB3_1: SPI[0/1] INT err:-110!
03:09:28.670183 INFO:    XTRX 1: Device closing
03:09:29.670112 INFO:    USB3380: IO thread termitaed with result 0
26.09.2018 03:09:29.670 (C) DeviceXTRXParams::open: cannot open device  /dev/xtrx0
26.09.2018 03:09:30.303 (I) WebAPIServer::start: starting web API server at http://127.0.0.1:8091
03:09:37.884526 INFO:    USB3380 dual fly GPEP RX mode is on, TX mode is on
03:09:37.884762 INFO:    USB3380: USB338x Speed: 480 Mbps
03:09:37.894569 INFO:    USB3380: Initialized PCIe dev 0100 [10ee:7012]
03:09:37.895783 INFO:    USB3380: IO thread started
03:09:37.896322 ERROR:   XTRX P: 00000066 00000000 <-> 00000000
03:09:37.896583 ERROR:   XTRX P: 00000066 00000000 <-> 00000000
03:09:37.896595 INFO:    XTRX USB3_1: RX DMA STOP MIMO (BLK:0 TS:0); TX DMA STOP MIMO
03:09:37.896684 INFO:    XTRX USB3_1: Device `/dev/xtrx0` was opened
03:09:38.009781 ERROR:   XTRX USB3_1: SPI[0/1] INT err:-110!
03:09:38.110158 ERROR:   XTRX USB3_1: SPI[0/1] INT err:-110!
03:09:38.210530 ERROR:   XTRX USB3_1: SPI[0/1] INT err:-110!
03:09:38.310923 ERROR:   XTRX USB3_1: SPI[0/1] INT err:-110!
03:09:38.411605 ERROR:   XTRX USB3_1: SPI[0/1] INT err:-110!
03:09:38.512336 ERROR:   XTRX USB3_1: SPI[0/1] INT err:-110!
03:09:38.613115 ERROR:   XTRX USB3_1: SPI[0/1] INT err:-110!
03:09:38.613227 INFO:    XTRX 1: Device closing
03:09:39.613127 INFO:    USB3380: IO thread termitaed with result 0
26.09.2018 03:09:39.613 (C) DeviceXTRXParams::open: cannot open device  /dev/xtrx0
Segmentation fault

f4exb, 2018-09-26:
Of course it is not. Just RTFM...
It is in the plans, I just wait for my unit for testing.

closed 2018-09-26 by f4exb

f4exb, 2018-09-26:
FYI: https://www.crowdsupply.com/fairwaves/xtrx/updates/shipping-testing-and-reworking
Will have to wait for the USB3 adapter.

hmne, 2018-09-26:
> FYI: https://www.crowdsupply.com/fairwaves/xtrx/updates/shipping-testing-and-reworking
> Will have to wait for the USB3 adapter.

hi thanks for your comment for your information
also i tried to mounting xtrx directly to motherboard pci not using usb3 adapter for my test 
and fresh install Kali Linux and Ubuntu and nothing worked for me

DG9BFC, 2019-02-01:
do you have your xtrx??
dg9bfc

f4exb, 2019-02-02:
Supported since v4.4.0