f4exb/sdrangel #221: Feature request: Support for SoapySDR.

ast, 2018-09-28:
Support for SoapySDR, an SoapyRemote in particular, would be really nice.

f4exb, 2018-09-28:
While the support of SoapySDR has been already invoked in tickets like this one: #105 and would be an interesting addition I don't see the point of SoapyRemote since you can operate SDRangel as a remote headless server more specifically since version 4.1.

The support of SoapySDR needs a bit of thinking and as I imagine as a very high level design breaks into two essential parts:
  - Addition of a SoapySDR virtual device so that it is seen by the application as just another device possibly with multiple input and output ports (like BladeRF v2 or LimeSDR USB)
  - Addition of a pair of plugins that can handle a stream of a virtual SoapySDR device. As far as I remember from looking at SoapySDR implementation details the streaming is muxed between Rx or Tx channels so that would be more something along the lines of the plugin for BladerRF v2 just being developed with a single thread to support the streaming of all channels.

idl0r, 2018-09-28:
I second this as it would add support for several devices and adding new ones might be even easier. Including remote device support etc.

f4exb, 2018-09-28:
At the risk of repeating myself you do arleady have remote device support. See:
  - https://github.com/f4exb/sdrangel/releases/tag/v4.1.0
  - https://github.com/f4exb/sdrangel/wiki/SDRangel-server
  - https://github.com/f4exb/sdrangel/wiki/SDRangel-server-with-supervisor-in-Arch-Linux (and the like for other disto flavors)

ast, 2018-09-30:
> The "server mode version" sdrangelsrv will not run on any ARM machine in particular the Raspberry Pi. It will work only on x86_64 architecture and you will need a server based on Intel Xeon E5 series or better.

Thank you for the feedback on my request! I was under the impression that the SDRangel server only runs on more powerful hardware, maybe this is incorrect? I'm not so experienced with SDRangel. The only advantage of having a SoapySDR plugin would be the possibility of using more narrowband SDRs remote with a Raspberry Pi or similar board.

We are in the process of developing a 96kHz bandwidth Pi based SDR transceiver, but with very high dynamic range so we were interested in adding support for this in SDRangel, and since we already have a Soapy module and use SoapyRemote with GQRX (without tx of course), it seemed like a good way with some additional advantages.

I will look into what it takes to implement this as a plugin. Perhaps I can just do it myself.

ast, 2018-09-30:
Ok, that's great news. I will look into writing an SDRangel plugin for our device.

>  I have had troubles with a RPi3B+ due to USB/Ethernet conflicts that I think have nothing to do with SDRangel.

I have also experienced this... it really only works for limited bandwidth, up to 192ksps has worked well for me. It's a shame they put ethernet and USB on the same bus. Our device use i2s and I have found that to work much better.

f4exb, 2018-10-31:
It appears that a pair of plugins (source and sink) is enough. SoapySDR acts as a wrapper with an arbitrary number of Rx and Tx channels and thus stream handling will be very much like BladeRF v.2. Allocating SoapySDR devices at top level is unnecessary.

f4exb, 2018-11-09:
Implemented  in v 4.3.0

f4exb, 2018-11-10:
Will be ...

erichahn, 2019-05-10:
I understand soapy is now supported, and UHD/USRP is supported under soapy. What is the status of getting a usrp b200 mini working under sdrangel at this point? I've been able to load the usrp b200 mini with cubicsdr utilizing soapy, but have not been successful with sdrangel.

f4exb, 2019-05-10:
I don't have a b200 mini so the status is unknown to me. You may try to find help from other users at the SDRangel discussion group: https://groups.io/g/sdrangel