f4exb/sdrangel #223: Crash in gf256.cpp

gvanem, 2018-10-03:
When activating the **Daemon channel sink** in `SDRangel.exe`, I got a crash in `gf256.cpp`.
I built everything using clang-cl v6 (the MSVC replacement) and with SSE3 enabled. 

The crash was caused due to destination was not aligned on 16-byte boundary in this instruction (from WinDbg):
```
eax=00000000 ebx=16cbd028 ecx=00000000 edx=00000000 esi=00000130 edi=00000010
eip=5f77a4d6 esp=1005d700 ebp=1005d75c iopl=0         nv up ei pl zr na pe nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00010246
daemonsink!gf256_ctx::gf256_muladd_mem_init+0xa6:
5f77a4d6 c5f829840b10070200 vmovaps xmmword ptr [ebx+ecx+20710h],xmm0 ds:002b:16cdd738=ff000000ff000000ff000000ff000000
```

Offending code:
```c
void gf256_ctx::gf256_muladd_mem_init()
{
    for (int y = 0; y < 256; ++y)
    {
        uint8_t lo[16], hi[16];

        // TABLE_LO_Y maps 0..15 to 8-bit partial product based on y.
        for (unsigned char x = 0; x < 16; ++x)
        {
            lo[x] = gf256_mul(x, static_cast<uint8_t>( y ));
            hi[x] = gf256_mul(x << 4, static_cast<uint8_t>( y ));
        }

        const GF256_M128 table_lo = _mm_set_epi8(
            lo[15], lo[14], lo[13], lo[12], lo[11], lo[10], lo[9], lo[8],
            lo[7], lo[6], lo[5], lo[4], lo[3], lo[2], lo[1], lo[0]);
        const GF256_M128 table_hi = _mm_set_epi8(
            hi[15], hi[14], hi[13], hi[12], hi[11], hi[10], hi[9], hi[8],
            hi[7], hi[6], hi[5], hi[4], hi[3], hi[2], hi[1], hi[0]);
! here: _mm_store_si128(MM256_TABLE_LO_Y + y, table_lo);
        _mm_store_si128(MM256_TABLE_HI_Y + y, table_hi);
    }
}
```

Seems the destination (`MM256_TABLE_LO_Y`) is not aligned on 16-byte boundary (as in gcc?). Simply adding this patch, fixed the crash:
```diff
--- a/gf256.h 2018-03-05 12:51:57
+++ b/gf256.h 2018-10-03 11:31:00
@@ -200,8 +200,8 @@
     // Muladd_mem tables
     // We require memory to be aligned since the SIMD instructions benefit from
     // aligned accesses to the MM256_* table data.
-    GF256_M128 MM256_TABLE_LO_Y[256];
-    GF256_M128 MM256_TABLE_HI_Y[256];
+    GF256_ALIGNED GF256_M128 MM256_TABLE_LO_Y[256];
+    GF256_ALIGNED GF256_M128 MM256_TABLE_HI_Y[256];

 private:
     int gf256_init_();
```

Does this make sense?

*PS* I fail to find any other use for ` GF256_ALIGNED`. Why not?

f4exb, 2018-10-03:
Code is actually in cm256cc. This seems to make sense.

gvanem, 2018-10-04:
Okay, wrong repo. My bad.

f4exb, 2018-10-04:
with gcc: opt/build/cm256cc/unit_test/../gf256.h:203:50: warning: ‘align’ attribute directive ignored [-Wattributes]

The align directive seems specific to clang-cl v6 is there any way to detect that it is used to compile?

f4exb, 2018-10-04:
In fact this should be: __attribute__((aligned(16))) not __attribute__((align(16)))
So I think your compiler picks up __declspec(align(16)) because _MSC_VER is defined

gvanem, 2018-10-05:
Isn't  `__attribute__((aligned(16)))` and `__declspec(align(16))` the same in clang-cl?

f4exb, 2018-10-05:
`__attribute__((align(16)))` would at least give you a warning this is why I think `__declspec(align(16))` was used.

f4exb, 2018-10-06:
Version 1.0.1 of cm256cc released with the fix. Build with this version will go in next release.