f4exb/sdrangel #224: Stream spectrum data for remote clients

f4exb, 2018-10-05:
It would be nice to be able to stream spectrum data across the network to a distant client so that it can process it for display or any other purpose. There are a few options but it should use UDP for transport and possibly RTP to cope with rate skew challenges. The exact details need to be defined.

f4exb, 2019-02-07:
Ideally this would be done via UDP as this is streaming data. It is not easy to find a way to communicate with browsers via UDP. However WebRTC seems to be designed for that purpose: https://www.html5rocks.com/en/tutorials/webrtc/datachannels/

f4exb, 2019-02-10:
Although not as efficient most web based SDR applications (OpenWebRX, ShinySDR at least) use web sockets to communicate with clients. Web sockets appear to be much more simple to implement.

f4exb, 2019-02-10:
Other resources for WebRTC:
  - http://research.edm.uhasselt.be/jori/page/Misc/QtWebRTC.html
  - https://sourcey.com/articles/webrtc-native-to-browser-video-streaming-example
  - https://sourcey.com/symple

How to retrieve and build WebRTC itself: https://webrtc.org/native-code/development/

f4exb, 2019-02-10:
One complete weekend trying to fiddle with WebRTC and nothing concrete: this is not the way to go.

f4exb, 2019-02-11:
Janus seems the most promising for setting up a WebRTC infrastructure. Note that it has to use https because browsers will not accept to use WebRTC over http unless running on localhost which is very limited.

f4exb, 2019-02-11:
Alas it seems that one has to go through the WebRTC code bloat to write clients to Janus or any WebRTC server in C++. So that's a no no...

f4exb, 2019-02-11:
Anything around trying to grab images from the existing spectrum display to make a video stream is a bit dodgy because it could have serious impact on display performance. Also it cannot work in environments without GL (server mode). See: https://doc.qt.io/qt-5/qopenglwidget.html#grabFramebuffer

It is comparatively easy to plug in (or replace in server mode) GLSpectrum::newSpectrum() method.

So back to a good old stream of bytes...

F5OEO, 2019-02-11:
Just FYI, an active WEBRTC implementation in golang (surely out of scope) : https://github.com/pions/webrtc

f4exb, 2019-02-11:
A bit off topic considering the golang inclination but interesting though. You will find a little bit of go with the script that reformats the .sdriq files broken  headers.

f4exb, 2020-05-02:
Eventually websockets are more convenient for browser integration and OK for relatively slow data. This is fairly easy to implement on server side with QtWebSockets.

f4exb, 2020-11-10:
To be retrofitted to master as v6

f4exb, 2020-11-12:
Deployed in v6.0.0

closed 2020-11-12 by f4exb