f4exb/sdrangel #226: Momentary Tx when changing Tx

KC7NOA, 2018-10-13:
I don’t know if this is a problem with SoapySdr or the SDR program i’m using … or Lime’s drivers for Linux …

I use LimeSDR-usb on Ubuntu 18.04 with SdrAngel and when ever i change frequencies with the Tx playing and mute on (the icon with the -120DB) showing … i get a momentary Tx of signal … i have not
used FM and no other mode …

Just wondering if any other FM SSB AM sdr programs show the same issue …

KC7NOA, 2018-10-15:
Lime people gave this as a posibility

https://discourse.myriadrf.org/t/momentary-tx-when-changing-frequencies/3488

f4exb, 2018-10-20:
Can you please describe in detail what you are trying to do? I cannot reproduce the issue with a simple scenario.

f4exb, 2018-10-28:
This is my FM modulator setup on Tx side with the Lime Mini:
[image]
And the settings for the Lime mini itself:
[image]
I have a RTL-SDR listening on the Rx side:
[image]
And when I move center frequency from 435000 kHz to 435010 and back this is what I see in the Rx waterfall:
[image]
Wideband is OK too:
[image]

So... no issues.

closed 2018-10-28 by f4exb