f4exb/sdrangel #229: pluto sink nfm no TX?

roelandjansen, 2018-10-18:
unless I really am doing something wrong:

I have rebuilt sdrangel on origin/master b397cd3a. 

If I use the pluto as sink and select am or nfm etc, shortly after I press 'play' a spike is transmitted and after that there is no output. The hw gain is set. A clear carrier made via a windows dvb-t works just fine. It happens here with opensuse leap 15, ubuntu 18.04 (virtual) as well as native in windows10.

The question is -- do I do stuff wrong or is this an issue?

[initial setup...]
2018-10-18 10:47:07.814 (I) WebAPIServer::start: starting web API server at http://127.0.0.1:8091
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=1: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute xo_correction=39999797: 9
DevicePlutoSDRBox::set_params: set attribute out_altvoltage1_TX_LO_frequency=435000000: 10
DevicePlutoSDRBox::set_params: set attribute out_voltage_rf_bandwidth=1500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage0_rf_port_select=A: 2
DevicePlutoSDRBox::set_params: set attribute out_voltage0_hardwaregain=-12.5: 6
DevicePlutoSDRBox::set_params: set attribute out_voltage0_hardwaregain=-26: 4
DevicePlutoSDRBox::set_params: set attribute out_voltage0_hardwaregain=-18: 4
DevicePlutoSDRBox::set_params: set attribute out_voltage0_hardwaregain=-10: 4
DevicePlutoSDRBox::set_params: set attribute out_voltage0_hardwaregain=-20.75: 7
DevicePlutoSDRBox::set_params: set attribute out_voltage0_hardwaregain=-22.75: 7
DevicePlutoSDRBox::set_params: set attribute out_voltage0_hardwaregain=-10: 4
DevicePlutoSDRBox::set_params: set attribute out_voltage0_hardwaregain=-1: 3
DevicePlutoSDRBox::set_params: set attribute out_voltage0_hardwaregain=0: 2
DevicePlutoSDRBox::set_params: set attribute out_voltage0_hardwaregain=-89.75: 7
DevicePlutoSDRBox::set_params: set attribute out_voltage0_hardwaregain=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=1: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_out_voltage_filter_fir_en=0: 2
DevicePlutoSDRBox::set_params: set attribute in_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage_sampling_frequency=2500000: 8
DevicePlutoSDRBox::set_params: set attribute xo_correction=39999797: 9
DevicePlutoSDRBox::set_params: set attribute out_altvoltage1_TX_LO_frequency=435000000: 10
DevicePlutoSDRBox::set_params: set attribute out_voltage_rf_bandwidth=1500000: 8
DevicePlutoSDRBox::set_params: set attribute out_voltage0_rf_port_select=A: 2
DevicePlutoSDRBox::set_params: set attribute out_voltage0_hardwaregain=0: 2

f4exb, 2018-10-18:
Do you set anything to modulate your NFM carrier: tone, audio input, morse code ? See: https://github.com/f4exb/sdrangel/blob/master/plugins/channeltx/modnfm/readme.md controls (10)

f4exb, 2018-10-21:
I did not find such an issue with the NFM modulator so it seems to be an usage question as mentioned above.

closed 2018-10-21 by f4exb