f4exb/sdrangel #231: Audio issue - Can't select devices ...

myotis6, 2018-10-23:
I intend to try WSJT-X with a LimeSDR (2X2 USB) using SDRAngel as the transciever.

SDRAngel basically works with the Lime when I initially set the 'radio' parameters. I get audio out the computer speakers by default.

To make WSJT-X work duplex, while having only one operating audio card, and wanting to avoid amplitude/phase distortion in analog audio chain processes, I decided to set up Pulseaudio with two vitual sinks. One for transmit and one for receive. I would send sound to a sink and use its monitor 'tap' to recover the audio crosswise in of each software vice-versa.

The SDRAngel Preferences_Audio dialog comes up as expected. It shows that SDRAngel is tying into ALSA directly for both in and out. The dialog shows other options which include "default", "System default device", the unused HDMI, and my two virtual sinks, similarly for both in and out. The problem is that selecting any of the options with the orange cursor bar does nothing. Selecting a null sink or the HDMI should kill the sound, but sound continues. Leaving the dialog and re-selecting it reveals that the choice has not actually changed when I 'OK'd out of the dialog. It is always selecting the ALSA. Running SDRAngel and selecting Pulse devices does not make SDRAngel appear at all in the application tabs of Pulseaudio's pavucontrol.

I tried changing the samplerate in the Sound dialog only to kill audio output all together. pavucontro sound meter bars show nothing being routed anywhere. Restoring the 44100 rate does not bring audio back at all. Neither does the "R" button, stopping and starting SDRAngel, nor the reload button in the Sample Devices control.

I did try patching audio via Jack, but neither Jack, SDRAngel, nor WSJT-X would see any available connections. Yes, I understand that a special Jack compatible driver hook is needed in the audio software.

My system is running Ubuntu 16.04 LTS on 4 fast cores. No VMs.

Would someone please help me troubleshoot the audio patching?

TIA, Wit N7PHD

f4exb, 2018-10-23:
The preferences > audio dialog deals with the audio devices themselves not the way they are connected. This interface has a specific documentation here: https://github.com/f4exb/sdrangel/blob/master/sdrgui/audio.md

To send audio data to one device or the other you have to select it from the channel demodulator plugin. For example with the SSB demod (the one you would use for WSJT) you have to right click on button 13: https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demodssb/readme.md

This will open a dialog that will let you choose the device for audio output with this particular channel demodulator. When more than one channel is connected to the same audio output the audio outputs of all these channels are mixed (this is what happens by default on the default device).

As this is a typical use case may be this deserves a page on the Wiki.

f4exb, 2018-11-25:
Audio wiki: https://github.com/f4exb/sdrangel/wiki/Audio-management

f4exb, 2018-11-25:
WSJT-X wiki: https://github.com/f4exb/sdrangel/wiki/WSJT-X

closed 2018-11-25 by f4exb