f4exb/sdrangel #237: Support for LimeSDR-mini

fury-pilot, 2018-11-08:
Hi/bonjour,

I've just upgraded from an RTL-SDR to a LimeSDR-mini (to enable transmit projects). I'd found indications of SDRangel support for LimeSDR but I cannot get SDRangel to work with the mini (Windows 7). Am I missing something? If not, do you have plans to add (Windows) support for the mini? I've tried a variety of SDR software and SDRangel is my favourite - and has transmit capability.

Thanks/cordialement,

Rob

f4exb, 2018-11-09:
This is a Windows only issue (hence changed the title). For some reason the Zadig drivers cannot work for the Lime mini. There is no way to circumvent this issue at the moment other than switching to a Linux O/S. I am currently exploring the possibility to use the MSVC 2017 toolchain instead of the Mingw32 one which could allow using different drivers but the outcome is still uncertain.

I still keep the issue open until the prospects of using MSVC are more definite.

fury-pilot, 2018-11-09:
Thankyou for explaining the cause of the problem. When I ran the Zadig driver install with my LimeSDR-mini installed, it timed out before completing the installation. It's been fine for the RTL-SDR.

I must admit, I still don't fully understand how Zadig, ExtIO and FTD3XX drivers fit together for LimeSDRs. I have managed to get a couple of general purpose SDR programs working with the LimeSDR-mini for reception, just using the FTD3XX drivers and the LimeSDR ExtIO. Does SDRangel work differently?

f4exb, 2018-11-09:
Yes SDRangel on Windows is based on the MinGW32 toolchain therefore it is not compatible with FTD3XX drivers which are MSVC based. The idea behind using a MSVC toolchain is to make it compatible with the LimeSuite version also based on MSVC and that in turn may be able to use the FTD3XX drivers. As you can see it is a fairly long way,

ExtIO is completely out of the picture. I am not planning to support anything related to ExtIO.

GkvJeep, 2018-11-10:
A question? why not support SoapySDR.
A good example of CubicSDR.... Lunix,Mac,Windows.

f4exb, 2018-11-10:
My bad, this issue was closed too early: https://github.com/f4exb/sdrangel/issues/221

Note that this will be Linux only at least in the first version. There are the same issues as with the native support.

SDRangel is targeted primarily at the Linux O/S and if possible extended to other OSes. Multi O/S is not a primary requirement.

f4exb, 2018-11-17:
The experimental (dev branch) build with MSVC makes the Lime mini work but it is not great. Maximum sample rate is around 3 to 4 MS/s. Above this mark there are too many packet drops and it is not even possible to change the sample rate while the application is running.

fury-pilot, 2018-11-17:
Is that version available to download for Windows? 

I'm not an expert Ubuntu user and so far have failed to get SDRangel (or indeed any SDR s/w) working. I'm using 16.04 LTS. Apologies this is off topic but grateful for any pointers.

f4exb, 2018-11-17:
Not yet. This has to be compiled from source using Qt Creator. But as said this is not great.

GkvJeep, 2018-11-18:
Hi https://github.com/f4exb
I put together a project for a visual studio to support SoapySDR without a QT creator.
It works only in the release version, but these are the problems of SoapySDR.
The speed normally works up to 10 MS/S.
Receiver is http://radioservice.ru/en/spectrum-jet-3-0-monitoring-receiver/
Problems only with the plugin chanalyzer, hangs.
A bit later lay out the screens.
Where to send the project?

f4exb, 2018-11-19:
Hi Konstantin,

looks good! This is what is supposed to be the magic with SoapySDR. As long as your hardware is supported by SoapySDR then it should be able to work with any SoapySDR application. I am not sure to understand the question "Where to send the project?" there should not be anything special to do with SDRangel. You probably need to talk to the SoapySDR guys so the support for your hardware can be integrated in the SoapySDR deployment.

There is an issue identified with the channel analyzer that shows up only on Windows and for relatively high speeds (~1.5 MS/s and above). I have never encountered such issue on Linux only that the trace memory function has issues coping with high speed which might be related.

Brgds.

f4exb, 2018-11-26:
Implemented in v4.3.0 with MSVC build

closed 2018-11-26 by f4exb