f4exb/sdrangel #238: Hackrf gives "Could not start sample source" on W10 and Ubuntu 18.04

Kees1957, 2018-11-10:
Hi, On both Ubuntu and W10 I get the error message couldn't start sample source. My Hackrf works on other SDR apps.

f4exb, 2018-11-10:
Any log ?

Kees1957, 2018-11-12:
2018-11-12 14:23:34.489 (I) SDRangel 4.2.4 Qt 5.11.1 32b x86_64 Windows 10 (10.0) DSP Rx:16b Tx:16b PID 18068

2018-11-12 14:23:41.548 (D) MainWindow::sampleSourceChanged: tab at 0

2018-11-12 14:23:41.548 (D) DeviceSourceAPI::saveSourceSettings: serializing source sdrangel.samplesource.filesource[0]: 

2018-11-12 14:23:41.548 (D) DSPDeviceSourceEngine::stopAcquistion

2018-11-12 14:23:41.548 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStop

2018-11-12 14:23:41.548 (D) DSPDeviceSourceEngine::gotoIdle

2018-11-12 14:23:41.548 (D) FileSourceInput::stop

2018-11-12 14:23:41.548 (D) DSPDeviceSourceEngine::setSourceSequence: seq: 0

2018-11-12 14:23:41.579 (C) HackRFInput::openDevice: could not open HackRF 457863c825571b1f

2018-11-12 14:23:41.579 (D) DSPDeviceSourceEngine::addSink:  FileRecord

2018-11-12 14:23:41.579 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAddBasebandSampleSink

2018-11-12 14:23:41.579 (D) FileRecord::handleMessage: DSPSignalNotification: m_inputSampleRate:  48000  m_centerFrequency:  435000000

2018-11-12 14:23:41.579 (D) DSPDeviceSourceEngine::setSource

2018-11-12 14:23:41.579 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPSetSource

2018-11-12 14:23:41.579 (D) DSPDeviceSourceEngine::gotoIdle

2018-11-12 14:23:41.579 (D) DSPDeviceSourceEngine::handleSetSource: set HackRF

2018-11-12 14:23:41.595 (D) MainWindow::setDeviceGUI: insert Rx tab at 0

2018-11-12 14:23:41.595 (D) MainWindow::setDeviceGUI: adding tab for HackRF[0] 457863c825571b1f

2018-11-12 14:23:41.595 (D) HackRFInput::applySettings

2018-11-12 14:23:41.595 (D) DSPDeviceSourceEngine::configureCorrections

2018-11-12 14:23:41.595 (D) DeviceSampleSource::calculateDeviceCenterFrequency:  desired center freq:  435000000  Hz  device center freq:  435000000  Hz  device sample rate:  2400000 S/s  Actual sample rate:  2400000 S/s  center freq position code:  2  image frequency:  435000000 Hz

2018-11-12 14:23:41.595 (D) FileRecord::handleMessage: DSPSignalNotification: m_inputSampleRate:  2400000  m_centerFrequency:  435000000

2018-11-12 14:23:41.595 (D) DSPDeviceSourceEngine::handleInputMessages

2018-11-12 14:23:41.595 (D) HackRFInput::applySettings:   m_centerFrequency:  435000000  Hz  m_LOppmTenths:  0  m_bandwidth:  1750000  m_lnaGain:  16  m_vgaGain:  16  m_log2Decim:  0  m_fcPos:  2  m_devSampleRate:  2400000  m_biasT:  false  m_lnaExt:  false  m_dcBlock:  false  m_linkTxFrequency:  false

2018-11-12 14:23:41.595 (D) DSPDeviceSourceEngine::handleInputMessages: message: DSPConfigureCorrection

2018-11-12 14:23:41.595 (D) DeviceSourceAPI::loadSourceSettings: Loading preset [default | no name]

2018-11-12 14:23:41.595 (D) DSPDeviceSourceEngine::handleInputMessages: message: DSPSignalNotification

2018-11-12 14:23:41.595 (D) Preset::findBestSourceConfig: serial matched (exact): id: sdrangel.samplesource.hackrf ser: 457863c825571b1f

2018-11-12 14:23:41.595 (D) DSPDeviceSourceEngine::handleInputMessages: DSPSignalNotification( 2400000 , 435000000 )

2018-11-12 14:23:41.595 (D) DeviceSourceAPI::loadSourceSettings: center frequency: 0 Hz

2018-11-12 14:23:41.595 (D) DSPDeviceSourceEngine::handleInputMessages: forward message to  SpectrumVis

2018-11-12 14:23:41.595 (D) DeviceSourceAPI::loadSourceSettings: deserializing source sdrangel.samplesource.hackrf[0]: 457863c825571b1f

2018-11-12 14:23:41.595 (D) DSPDeviceSourceEngine::handleInputMessages: forward message to  FileRecord

2018-11-12 14:23:41.595 (D) FileRecord::handleMessage: DSPSignalNotification: m_inputSampleRate:  2400000  m_centerFrequency:  435000000

2018-11-12 14:23:41.595 (D) DSPDeviceSourceEngine::handleInputMessages: DSPSignalNotification: guiMessageQueue: 0x138990f4

[... 67 lines cut ...]

2018-11-12 14:23:52.869 (D) DSPDeviceSourceEngine::gotoRunning:  HackRF  started

2018-11-12 14:23:52.869 (D) DSPDeviceSourceEngine::gotoError:  "Could not start sample source"

2018-11-12 14:23:52.869 (D) HackRFGui::handleInputMessages: message: DSPSignalNotification

2018-11-12 14:23:52.869 (D) HackRFGui::handleInputMessages: DSPSignalNotification: SampleRate:2400000, CenterFrequency:0

2018-11-12 14:23:52.869 (D) GLSpectrumGUI::handleInputMessages: message: GLSpectrum::MsgReportSampleRate

2018-11-12 14:23:53.100 (D) DSPDeviceSourceEngine::errorMessage

2018-11-12 14:23:53.100 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPGetErrorMessage

2018-11-12 14:26:30.936 (D) LoggingDialog::on_showFileDialog_clicked: selected: C:/Program Files/sdrangel/test

2018-11-12 14:26:33.090 (I) SDRangel 4.2.4 Qt 5.11.1 32b x86_64 Windows 10 (10.0) DSP Rx:16b Tx:16b PID 18068

 

Met vriendelijke groet,
Kees van der Wal
+31654320919
 <http://www.linkedin.com/in/cvanderwal> http://www.linkedin.com/in/cvanderwal

f4exb, 2018-11-13:
The error is here:
`2018-11-12 14:23:41.579 (C) HackRFInput::openDevice: could not open HackRF 457863c825571b1f`

Unfortunately the return code that can be found here: https://github.com/mossmann/hackrf/blob/master/host/libhackrf/src/hackrf.h is not reported in the error message. This can be enhanced.

Kees1957, 2018-11-13:
Thanks, I found the problem. The firmware was old, I updated the firmware and that solved the problem.

Kees1957, 2018-11-26:
ThxMet vriendelijke groet,Kees van der Wal+31654320919http://www.linkedin.com/in/cvanderwal

[
{

gaspar1987, 2019-08-26:
Could not start sample source