f4exb/sdrangel #239: No source available (when I manage to run it)

i-am-tof, 2018-11-11:
Hello,
I've got an up to date Ubuntu 16.04 Xenial amd64.
I can use a hackrf or a rtlsrd dongle with gqrx and many other programs without any problem.

Wondering using sdrangel, I installed it from "sdrangel_4.2.4-1_amd64_xenial.deb" but there was no way to run it. The sdrangel binary is not seen from command line.

So I installed it from github sources  (cmake .., make, make install). Everything goes fine. Then when I ran sdrangel, but it was unable to open these libraries despite there where present :
    /usr/local/lib/libsdrgui.so
    /usr/local/lib/liblogging.so
    /usr/local/lib/libsdrbase.so
    /usr/local/lib/libswagger.so
    /usr/local/lib/libhttpserver.so
    /usr/local/lib/libqrtplib.so
    /usr/local/lib/plugins/samplesink/liboutputhackrf.so
    /usr/local/lib/plugins/samplesource/libinputhackrf.so
I linked them in /usr/lib/, than sdrangel was able to load them, except libinputhackrf.so and liboutputhackrf.so.

Hackrf libs and rtlsdr dongle libs where installed from package manager prior to sdrangel.

I tried to link /usr/lib/samplesink to /usr/local/lib/plugins/samplesink and /usr/lib/samplesource to /usr/local/lib/plugins/samplesource, then sdrangel launched without loading error, but I didn't have any source available in the app.

There was no menu link to sdrangel, nor any icon installed, I had to put some icons by hand and create a launcher.
[image]

I removed all links, made a "make clean", erased the build directory and made a new "cmake ../make/make install". Everything compiled and installed without any error but sdrangel still do the same as before, complaining about the libs it can't open and refusing to launch.

I don't know if some cleaning is necessary before compiling it again. I'm stuck and I don't know how to do. Any help would be nice !

i-am-tof, 2018-11-12:
Hello,
Thanks for your response, not being a developer, I really appreciate such informations.
I spent much time trying to see sdrangel working. Doing so I red some details about how you compiled yours, but I was unable to retrieve the good page, so yes it would be useful to put these details as a dedicated readme file on the git.

I cleaned what I did (removed links, make uninstall, check everything) then retried with the deb package. No more warning about libs but unfortunately, I still have the same issue...no source in sdrangel.

[image]

Here is the sdrangel's terminal output, and the loaded plugins window, giving versions in case of.

$ 2018-11-12 07:10:51.862 (I) SDRangel 4.2.4 Qt 5.5.1 64b x86_64 Ubuntu 16.04.5 LTS DSP Rx:24b Tx:16b PID 12016
2018-11-12 07:10:51.862 (D) AudioDeviceManager::AudioDeviceManager: scan input devices
2018-11-12 07:10:51.976 (D) AudioDeviceManager::AudioDeviceManager: input device #0: HDA ATI SB
2018-11-12 07:10:51.976 (D) AudioDeviceManager::AudioDeviceManager: input device #1: HDA NVidia
2018-11-12 07:10:51.976 (D) AudioDeviceManager::AudioDeviceManager: input device #2: default
2018-11-12 07:10:51.976 (D) AudioDeviceManager::AudioDeviceManager: input device #3: alsa_output.pci-0000_00_14.2.analog-stereo.monitor
2018-11-12 07:10:51.976 (D) AudioDeviceManager::AudioDeviceManager: scan output devices
2018-11-12 07:10:52.059 (D) AudioDeviceManager::AudioDeviceManager: output device #0: HDA ATI SB
2018-11-12 07:10:52.059 (D) AudioDeviceManager::AudioDeviceManager: output device #1: HDA NVidia
2018-11-12 07:10:52.059 (D) AudioDeviceManager::AudioDeviceManager: output device #2: default
2018-11-12 07:10:52.059 (D) AudioDeviceManager::AudioDeviceManager: output device #3: alsa_output.pci-0000_00_14.2.analog-stereo
2018-11-12 07:10:52.059 (D) MainWindow::MainWindow: start
2018-11-12 07:10:52.142 (D) MainWindow::MainWindow: load settings...
2018-11-12 07:10:52.142 (D) MainWindow::loadSettings
2018-11-12 07:10:52.142 (D) AudioDeviceManager::deserialize
2018-11-12 07:10:52.142 (D) AudioDeviceManager::debugAudioOutputInfos:  name:  "System default device"  sampleRate:  48000  udpAddress:  "127.0.0.1"  udpPort:  9998  copyToUDP:  false  udpUseRTP:  false  udpChannelMode:  0
2018-11-12 07:10:52.142 (D) MainWindow::MainWindow: load plugins...
2018-11-12 07:10:52.142 (D) PluginManager::loadPlugins:  /opt/sdrangel/bin/../lib/plugins , /opt/sdrangel/bin/plugins
2018-11-12 07:10:52.142 (D) PluginManager::loadPluginsDir: fileName:  libchanalyzer.so
2018-11-12 07:10:52.143 (I) PluginManager::loadPluginsDir: loaded plugin libchanalyzer.so
2018-11-12 07:10:52.143 (D) PluginManager::loadPluginsDir: fileName:  libdaemonsink.so
2018-11-12 07:10:52.143 (I) PluginManager::loadPluginsDir: loaded plugin libdaemonsink.so
2018-11-12 07:10:52.143 (D) PluginManager::loadPluginsDir: fileName:  libdemodam.so
2018-11-12 07:10:52.144 (I) PluginManager::loadPluginsDir: loaded plugin libdemodam.so
2018-11-12 07:10:52.144 (D) PluginManager::loadPluginsDir: fileName:  libdemodatv.so
2018-11-12 07:10:52.145 (I) PluginManager::loadPluginsDir: loaded plugin libdemodatv.so
2018-11-12 07:10:52.145 (D) PluginManager::loadPluginsDir: fileName:  libdemodbfm.so
2018-11-12 07:10:52.146 (I) PluginManager::loadPluginsDir: loaded plugin libdemodbfm.so
2018-11-12 07:10:52.146 (D) PluginManager::loadPluginsDir: fileName:  libdemoddsd.so
2018-11-12 07:10:52.147 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddsd.so
2018-11-12 07:10:52.147 (D) PluginManager::loadPluginsDir: fileName:  libdemodlora.so
2018-11-12 07:10:52.147 (I) PluginManager::loadPluginsDir: loaded plugin libdemodlora.so
2018-11-12 07:10:52.147 (D) PluginManager::loadPluginsDir: fileName:  libdemodnfm.so
2018-11-12 07:10:52.148 (I) PluginManager::loadPluginsDir: loaded plugin libdemodnfm.so
2018-11-12 07:10:52.148 (D) PluginManager::loadPluginsDir: fileName:  libdemodssb.so
2018-11-12 07:10:52.149 (I) PluginManager::loadPluginsDir: loaded plugin libdemodssb.so
2018-11-12 07:10:52.149 (D) PluginManager::loadPluginsDir: fileName:  libdemodwfm.so
2018-11-12 07:10:52.149 (I) PluginManager::loadPluginsDir: loaded plugin libdemodwfm.so
2018-11-12 07:10:52.149 (D) PluginManager::loadPluginsDir: fileName:  libudpsink.so
2018-11-12 07:10:52.150 (I) PluginManager::loadPluginsDir: loaded plugin libudpsink.so
2018-11-12 07:10:52.150 (D) PluginManager::loadPluginsDir: fileName:  libdaemonsource.so
2018-11-12 07:10:52.151 (I) PluginManager::loadPluginsDir: loaded plugin libdaemonsource.so
2018-11-12 07:10:52.151 (D) PluginManager::loadPluginsDir: fileName:  libmodam.so
2018-11-12 07:10:52.151 (I) PluginManager::loadPluginsDir: loaded plugin libmodam.so
2018-11-12 07:10:52.151 (D) PluginManager::loadPluginsDir: fileName:  libmodatv.so
2018-11-12 07:10:52.182 (I) PluginManager::loadPluginsDir: loaded plugin libmodatv.so
2018-11-12 07:10:52.182 (D) PluginManager::loadPluginsDir: fileName:  libmodnfm.so
2018-11-12 07:10:52.183 (I) PluginManager::loadPluginsDir: loaded plugin libmodnfm.so
2018-11-12 07:10:52.183 (D) PluginManager::loadPluginsDir: fileName:  libmodssb.so
2018-11-12 07:10:52.183 (I) PluginManager::loadPluginsDir: loaded plugin libmodssb.so
2018-11-12 07:10:52.183 (D) PluginManager::loadPluginsDir: fileName:  libmodwfm.so
2018-11-12 07:10:52.184 (I) PluginManager::loadPluginsDir: loaded plugin libmodwfm.so
2018-11-12 07:10:52.184 (D) PluginManager::loadPluginsDir: fileName:  libudpsource.so
[... cut ...]
DSPDeviceSourceEngine::removeThreadedSink:  ThreadedBasebandSampleSink(DownChannelizer(WFMDemod))
DSPDeviceSourceEngine::handleSynchronousMessages:  DSPRemoveThreadedBasebandSampleSink
ThreadedBasebandSampleSink::stop
MessageQueue::~MessageQueue: message:  DSPConfigureChannelizer  was still in queue
MessageQueue::~MessageQueue: message:  DSPConfigureChannelizer  was still in queue
SDRangel quit.
DSPDeviceSourceEngine::stop
DSPDeviceSourceEngine::gotoIdle
DVSerialEngine::release

[image]

Would you have an idea ?

f4exb, 2018-11-12:
It happens here:
2018-11-12 07:10:57.723 (W) FileSourceInput::start: file not open. not starting
2018-11-12 07:10:57.723 (D) DSPDeviceSourceEngine::gotoError: "Could not start sample source"

You have selected the File Source plugin but did not specify any file to play.

However I don't think using the File Source is what you really want to do so at the right where it says "File Source" there is a hand with pointing finger icon that you need to press to open the device selection dialog. There you should be able to select HackRF from the list (I think you have a HackRF connected).

i-am-tof, 2018-11-12:
Hello

I'm ashamed...It works.
There is so much buttons, functions, options, I'm completely lost.
I have to learn using this marvel. Sdrangel has nothing to do with any other.

**Thanks a lot** for this jewel, and for your help too.

closed 2018-11-12 by i-am-tof