f4exb/sdrangel #240: suggestion: Channels dock to window

sgoudelis, 2018-11-12:
I think for people that use more than on monitors (like me) it would be fantastic if the channel window/view would actually pop out and became an independent window that would be fit in another monitor, and not disappear when the SDRAngel app looses focus as it is now.

f4exb, 2018-11-13:
This all depends on Qt. I am not sure this is feasible.

sgoudelis, 2018-11-13:
Yes that is the main issue. Also it would be nice if it was an actual window that is managed by the window manager.

f4exb, 2018-11-13:
I don't think this is possible. All docking widgets belong to the main window. Nothing prevents you from working on another application on the top of the main window though.

closed 2018-11-25 by sgoudelis