f4exb/sdrangel #241: Presets storing

i-am-tof, 2018-11-15:
Hello,

A backup function would be a great addition to save all groups and presets at once.
I would know where the "save the current settings (inc. presets)" button store its result.
I use sdrangel v4.2.4 on Ubuntu Xenial 16.04.

i-am-tof, 2018-11-15:
I saw for one preset at a time, but sadly it doesn't save belonging group.
Rebuilding presets manually will become more and more boring with time, and I especially need to backup and to replicate, just as you told.

Awaiting this enhancement, in which file(s) does sdrangel stores its settings+presets ?

f4exb, 2018-11-15:
The missing belonging group is intentional because when you export and import presets one at a time you do not necessarily want to recreate them in the same group.

Presets use the Qt preferences mechanism for storage so this is system dependent. On Linux systems this is in `~/.config/f4exb/SDRangel.conf` data is binary base64 encoded so although it is ASCII this is effectively humanly unreadable.

beta-tester, 2019-01-16:
> On Linux systems this is in ~/.config/f4exb/SDRangel.conf data is binary base64 encoded so although it is ASCII this is effectively humanly unreadable.

where is it stored on a windows machine?

f4exb, 2019-01-17:
I've found a way to retrieve it with Qt interface: http://doc.qt.io/qt-5/qsettings.html#fileName Since this is a recurring question I will add a message in the log and in the about panel also.

For readibility and edit of the presets maybe the best option is to implement it via REST API. At the moment interaction with presets using REST API is rudimentary.

f4exb, 2019-01-17:
As you mentioned on Windows this is stored in the registry and I think that the regedit utility can export and import keys:

[image]

beta-tester, 2019-01-17:
yes, now i know that next time i will export the registry key on windows. ;-)
but that way you can not transfere presets from windows to linux or from linux to windows.
anyhow, first i have to go the long way to restore all my frequency presets...
hopefully i can find all the frequencies i had in the past...
how was it... no backup, no pity... :-D

f4exb, 2019-01-17:
Going through REST API you could even transfer live between two running applications over the network.

beta-tester, 2019-01-17:
Thank you, that despite my obtuseness, you still have not given up to help me ...
i just found the REST API with its documentation in SDRangel you mentioned.
http://127.0.0.1:8091 ...
oops, how could i overlook its existence?
now i have to study you well documeted REST API of SDRangel.
there are also examples for different programming languages... very nice.

f4exb, 2019-01-17:
You will see that there is not much yet for the presets: just transposing what you can do with the GUI.  The purpose of implementing this ticket would be to enrich it so as to be able to deal with presets and settings generally in JSON format.

f4exb, 2019-08-14:
Implemented in v4.11.6: /sdrangel/config with GET, PUT, PATCH methods

closed 2019-08-14 by f4exb