f4exb/sdrangel #242: Windows: MSVC2017 build

f4exb, 2018-11-16:
It is possible to use the MSVC2017 with Qt. The MSVC build environment can be downloaded for free from Microsoft there's no need to use Visual Studio.

This could provide more support from 3rd party libraries that exist only (or easier to deal with) with MSVC linkage. Most noticeably SoapySDR and its support libraries.

GkvJeep, 2018-11-19:
Hi !
The positive effect was obtained after changing value timeoutUs to the default .(LINE 132)
Also changed the error handling (LINE:145-149)
[img]https://c.radikal.ru/c39/1811/ab/c84aaac722aet.jpg[/img][/url]
Konstantin.

f4exb, 2018-11-19:
This is more related to SoapySDR support. I see that SOAPY_SDR_OVERFLOW can be accepted. For the timeout I have changed it back to actually 10 times the block length in time and it works fine with the LimeSDR at least up to 26 MS/s.

GkvJeep, 2018-11-22:
HI ! 
An error was fixed in the file FindFFmpeg.cmake branch dev.
Line 140, bad name POSTPROCESS, must be POSTPROC.
Konstantin.