f4exb/sdrangel #243: Crash at Windows

OnlySlon, 2018-11-16:
Windows 10 64bit + Adalm-pluto
Sdrangel 4.2.4  works 0-5 min then crashes
0x6E590960 (sdrgui.dll)  sdrangel.exe: 0xC0000005: 
write address 0x100FE148.
Looks like it happends inside ScopeVis::processTraces

f4exb, 2018-11-17:
Normally this kind of error translates to "segmentation fault" on Linux systems but I was unable to reproduce by trivial means. At least at first the issue does not seem to be in the code so I will not be able to fix it. Can you please make a screenshot of your channel analyzer display right before it crashes so I can get an idea about what are the parameters you are using?

f4exb, 2018-11-17:
On Windows It is indeed fairly easy to make it crash by playing with the trace length handle. I cannot derive a precise condition on when it would fail though. It seems much more difficult to make it crash if the trace is long enough say about 1/4 of the handle slide length and above. 

There must be a memory allocation issue but not showing up on Linux where I would be able to analyze the core dump.

Note that this happens with the MSVC 2017 (64 bit)  version in development also.

f4exb, 2018-11-19:
It looks like a race condition between the glScope that reads the trace buffer and the ScopeVis that writes to it. There is a double buffer mechanism but that's not enough when writing speed increases. What maybe happens is that Linux tolerates write and read competition and Windows does not. In any case the trace display is affected.

GkvJeep, 2018-11-20:
The guys from  CubicSDR have  beautiful streaming class 
[https://github.com/cjcliffe/CubicSDR/blob/master/src/util/ThreadBlockingQueue.h]
Can try on its base?

f4exb, 2018-11-20:
Mutexes are not that great and Qt can do things differently. Please check: https://woboq.com/blog/introduction-to-lockfree-programming.html

GkvJeep, 2018-11-20:
It may help.
stack 
line 263
line 426
line 739
exception line 645
[image]

GkvJeep, 2018-11-22:
Well done ;-)
Works great!