f4exb/sdrangel #245: Analyzer NG not work.

bdn76, 2018-11-18:
2018-11-18 14:12:57.481 (I) SDRangel 4.3.0 Qt 5.11.2 64b x86_64 Debian GNU/Linux buster/sid DSP Rx:16b Tx:16b PID 2664
2018-11-18 14:12:58.005 (I) PluginManager::loadPluginsDir: loaded plugin libchanalyzer.so
2018-11-18 14:12:58.006 (I) PluginManager::loadPluginsDir: loaded plugin libchanalyzerng.so
2018-11-18 14:12:58.007 (I) PluginManager::loadPluginsDir: loaded plugin libdemodam.so
2018-11-18 14:12:58.008 (I) PluginManager::loadPluginsDir: loaded plugin libdemodatv.so
2018-11-18 14:12:58.009 (I) PluginManager::loadPluginsDir: loaded plugin libdemodbfm.so
2018-11-18 14:12:58.064 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddatv.so
2018-11-18 14:12:58.065 (I) PluginManager::loadPluginsDir: loaded plugin libdemodlora.so
2018-11-18 14:12:58.066 (I) PluginManager::loadPluginsDir: loaded plugin libdemodnfm.so
2018-11-18 14:12:58.067 (I) PluginManager::loadPluginsDir: loaded plugin libdemodssb.so
2018-11-18 14:12:58.068 (I) PluginManager::loadPluginsDir: loaded plugin libdemodtcpsrc.so
2018-11-18 14:12:58.069 (I) PluginManager::loadPluginsDir: loaded plugin libdemodudpsrc.so
2018-11-18 14:12:58.069 (I) PluginManager::loadPluginsDir: loaded plugin libdemodwfm.so
2018-11-18 14:12:58.070 (I) PluginManager::loadPluginsDir: loaded plugin libudpsink.so
2018-11-18 14:12:58.071 (I) PluginManager::loadPluginsDir: loaded plugin libmodam.so
2018-11-18 14:12:58.720 (I) PluginManager::loadPluginsDir: loaded plugin libmodatv.so
2018-11-18 14:12:58.722 (I) PluginManager::loadPluginsDir: loaded plugin libmodnfm.so
2018-11-18 14:12:58.723 (I) PluginManager::loadPluginsDir: loaded plugin libmodssb.so
2018-11-18 14:12:58.724 (I) PluginManager::loadPluginsDir: loaded plugin libmodudpsink.so
2018-11-18 14:12:58.725 (I) PluginManager::loadPluginsDir: loaded plugin libmodwfm.so
2018-11-18 14:12:58.726 (I) PluginManager::loadPluginsDir: loaded plugin libudpsource.so
2018-11-18 14:12:58.727 (I) PluginManager::loadPluginsDir: loaded plugin liboutputbladerf.so
2018-11-18 14:12:58.728 (I) PluginManager::loadPluginsDir: loaded plugin liboutputfilesink.so
2018-11-18 14:12:58.730 (I) PluginManager::loadPluginsDir: loaded plugin liboutputhackrf.so
2018-11-18 14:12:58.732 (I) PluginManager::loadPluginsDir: loaded plugin liboutputsoapysdr.so
2018-11-18 14:12:58.733 (I) PluginManager::loadPluginsDir: loaded plugin libinputairspy.so
2018-11-18 14:12:58.734 (I) PluginManager::loadPluginsDir: loaded plugin libinputbladerf.so
2018-11-18 14:12:58.735 (I) PluginManager::loadPluginsDir: loaded plugin libinputfcdpro.so
2018-11-18 14:12:58.737 (I) PluginManager::loadPluginsDir: loaded plugin libinputfcdproplus.so
2018-11-18 14:12:58.738 (I) PluginManager::loadPluginsDir: loaded plugin libinputfilesource.so
2018-11-18 14:12:58.739 (I) PluginManager::loadPluginsDir: loaded plugin libinputhackrf.so
2018-11-18 14:12:58.740 (I) PluginManager::loadPluginsDir: loaded plugin libinputrtlsdr.so
2018-11-18 14:12:58.741 (I) PluginManager::loadPluginsDir: loaded plugin libinputsoapysdr.so
2018-11-18 14:12:58.742 (I) PluginManager::loadPluginsDir: loaded plugin libinputtestsource.so
[INFO] Opening HackRF One #0 909864c837659dcf...
2018-11-18 14:12:59.678 (W) HackRFInput::setDeviceCenterFrequency: frequency set to 435000000 Hz
2018-11-18 14:13:01.275 (I) WebAPIServer::start: starting web API server at http://127.0.0.1:8091
2018-11-18 14:13:01.296 (W) HackRFInput::setDeviceCenterFrequency: frequency set to 435000000 Hz
sdrangel: symbol lookup error: /usr/local/lib/plugins/channelrx/libchanalyzerng.so: undefined symbol: _ZN9GLScopeNGC1EP7QWidget

f4exb, 2018-11-18:
4.3.0 is on dev branch so I suppose you compile from source

A few things:
- please do not install in system paths which  usually is the default with cmake. This is a bad idea. Install in say `/opt/install/sdrangel` this way you can wipe out the entire directory which is what you should do before install (see next)
- start with a clean installation libchanalyzerng.so has been discontinued for a while so I suppose it is there from a previous deployment

f4exb, 2018-11-19:
Corrected the title with the proper issue.

bdn76, 2018-11-19:
I'm compile new one to /opt/sdrangel and can not found analyzer NG plugin but found it in old dest:

$ find /opt/sdrangel -name libchanalyzerng.so
$ find /usr/local -name libchanalyzerng.so
/usr/local/lib/plugins/channelrx/libchanalyzerng.so
$ 

I'm think this is relict from old install and need to be cleaned:
$ ls -l /usr/local/lib/plugins/channelrx/libchanalyzer*
-rw-r--r-- 1 root staff 190216 Nov 19 08:34 /usr/local/lib/plugins/channelrx/libchanalyzer.so
-rw-r--r-- 1 root staff 159512 Apr 25  2018 /usr/local/lib/plugins/channelrx/libchanalyzerng.so

Am i right?

f4exb, 2018-11-19:
yes you can (have to in fact) get rid of `libchanalyzerng.so`

closed 2018-11-19 by bdn76