f4exb/sdrangel #250: Groups arrangement

i-am-tof, 2018-11-25:
Hello

Each time a preset is created or modified, all groups become developed.
It would be great to keep groups state as it was set.
I noticed that sdrangel takes a little time to refresh that list when doing so. Perhaps modifying that could cut this tiny delay.

f4exb, 2018-11-25:
If the list is long then it takes some time to re-arrange everything. Not sure it is possible to optimize this as it depends very much on the underlying structures managed by Qt. 

There is a sorting order by group then frequency that helps finding things out. I am not planning to change that nor to spend time in the collapse/expand thing.

f4exb, 2019-01-20:
No changes planned,

closed 2019-01-20 by f4exb