f4exb/sdrangel #251: Cmake VS2017

GkvJeep, 2018-11-25:
Hi Edouard.
In the attached file, the changes for assembly through Cmake for VS2017, without *.pro.
[ChangeCmakeVS.txt](https://github.com/f4exb/sdrangel/files/2613291/ChangeCmakeVS.txt)
Konstantin.

f4exb, 2018-11-26:
Do you still use some kind of tool like cmake GUI ?

GkvJeep, 2018-11-26:
Yes Cmake GUI. 

Prebuilt windows binaries.
QT5         (must match MSVC version) - http://qt-project.org/downloads
PothosSDR    http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D
Boots        https://sourceforge.net/projects/boost/files/boost-binaries/

GkvJeep, 2018-11-27:
Hi . 
What version of FFMPEG are you using for assembly?
3.2 .... 4.x  
I do the assembly under Windows https://ffmpeg.zeranoe.com/builds/win64/dev/

f4exb, 2018-11-27:
This is used in the DATV demodulator that is not present in Windows distributions.

GkvJeep, 2018-11-27:
I want to help you do it.

f4exb, 2018-11-27:
I see you managed to compile it on MSVC well done. If I take the build on Ubuntu 18.04 as a reference this is ffmpeg version 3.4.2-2

GkvJeep, 2018-11-27:
Changes
//===============  datvconstellation.h  =============================
sdrangel-4.3.0\plugins\channelrx\demoddatv\datvconstellation.h

#ifdef _WINDOWS
#include <sys/timeb.h>
#else
#include <sys/time.h>
#endif
//=========== leansdr\generic.h  =================================
sdrangel-4.3.0\plugins\channelrx\demoddatv\leansdr\generic.h
#ifdef _WINDOWS
#include <stdlib.h>
#include <io.h>
#else
#include <unistd.h>
#endif
//=============  leansdr\dvb.h ===============================
sdrangel-4.3.0\plugins\channelrx\demoddatv\leansdr\dvb.h
line 672 comment FECxx
{ //[FEC12] =
{ 1, 2, polys_fec12 },// [FEC23] =
{ 2, 3, polys_fec23 },// [FEC46] =
{ 4, 6, polys_fec46 },// [FEC34] =
{ 3, 4, polys_fec34 },// [FEC56] =
{ 5, 6, polys_fec56 },// [FEC78] =
{ 7, 8, polys_fec78 },// [FEC45] =

// ========  Fixed error  =================
sdrangel-4.3.0\plugins\channelrx\demoddatv\datvideostream.cpp
 Fixed error 
 Line 52 m_objMutex.unlock();
 Сall unlock() witout  lock() from the DATVideostream::DATVideostream()

f4exb, 2018-11-27:
This is a Qt resources file that is built from a .qrc file. See this: 
https://github.com/f4exb/sdrangel/blob/bc7ad10f9d9a7694320e8ecebab65eb8b9c1b5e7/CMakeLists.txt#L244

This is actually the internal static web pages and the swagger-ui and is essential.

f4exb, 2018-11-28:
Why do you comment all [FECxx] in plugins\channelrx\demoddatv\leansdr\dvb.h ?
This is a breaking code change.

GkvJeep, 2018-11-28:
Otherwise compile errors C3480,C3260,C2143.
There is no code violation.

f4exb, 2018-11-28:
OK, i'll test it. I am now able to build DATV Demod with .pro file. Using dev and shared distribution of ffmpeg from here: https://ffmpeg.zeranoe.com/builds/ (same as yours but selects the most up to date and you need the shared for the .dlls distribution).

GkvJeep, 2018-11-28:
I make through cmake GUI.
Secondly, you can select previous versions there,dev and  shared.
Label Previous Builds and three buttons .... Windows 64-bit build ... etc

GkvJeep, 2018-11-30:
Hi !
A little bit of this.
https://github.com/f4exb/sdrangel/tree/master/plugins/channelrx/demoddsd#dv-serial-device-support
" Note also that this is not supported in Windows because of trouble with COM port support (contributors welcome!).
What help is needed?

f4exb, 2018-11-30:
Well I have no idea why it does not work in Windows. This is a serial port over USB interface so I would expect it to create a virtual COM port. I could never get it working on Windows but works like a charm on Linux and I use it every day for digital voice as it gives a superior sound.

GkvJeep, 2018-12-01:
Not Windows problem.
page 7.
3.2.
FTDI's R, C and BM series chips allow the latency timer to be changed from 16 milliseconds to any
value from 1 to 255 milliseconds, in 1 millisecond increments. When using the FTDI Virtual COM
Port driver the latency timer can be set in the port properties page. In Windows, the port properties
page is accessed via the Control Panel > System > Device Manager. For Windows 2000 and XP,
the initial value of the latency timer can also be pre-configured in ftdiport.inf by changing the value
of the last number in the following line:
[FtdiPort.NT.HW.AddReg]
HKR,,"LatencyTimer",0x00010001,16
where again, 16 milliseconds is the default value.
When using FTDI's D2XX direct driver the function FT_SetLatencyTimer can be used to adjust the
value of the latency timer.

https://www.ftdichip.com/Support/Documents/AppNotes/AN232B-04_DataLatencyFlow.pdf

f4exb, 2018-12-01:
This is a Windows + AMBE dongle issue and is a different problem that should be handled in another issue. This requires specific code for Windows.

GkvJeep, 2018-12-02:
Checked.
No comment required, line 52 files \sdrgui\CMakeLists.txt.
Otherwise, a link error is fixed.

GkvJeep, 2018-12-03:
Wish, add option  ;-)
option(BUILD_PLAGINSRV "Build PLAGINS SRV" OFF)
........
if (BUILD_PLAGINSRV)
	add_subdirectory(pluginssrv)
endif()

GkvJeep, 2018-12-03:
PVS-studio fixed.

V773 Visibility scope of the 'fifos' pointer was exited without releasing the memory. A memory leak is possible. soapysdrinput.cpp 522
V773 Visibility scope of the 'log2Decims' pointer was exited without releasing the memory. A memory leak is possible. soapysdrinput.cpp 522
V773 Visibility scope of the 'fcPoss' pointer was exited without releasing the memory. A memory leak is possible. soapysdrinput.cpp 522

V773 Visibility scope of the 'fifos' pointer was exited without releasing the memory. A memory leak is possible. soapysdrinput.cpp 656
V773 Visibility scope of the 'log2Decims' pointer was exited without releasing the memory. A memory leak is possible. soapysdrinput.cpp 656
V773 Visibility scope of the 'fcPoss' pointer was exited without releasing the memory. A memory leak is possible. soapysdrinput.cpp 656

V773 Visibility scope of the 'fifos' pointer was exited without releasing the memory. A memory leak is possible. soapysdroutput.cpp 486
V773 Visibility scope of the 'log2Interps' pointer was exited without releasing the memory. A memory leak is possible. soapysdroutput.cpp 486

V774 The 'soapySDROutputThread' pointer was used after the memory was released. soapysdroutput.cpp 610

V773 Visibility scope of the 'fifos' pointer was exited without releasing the memory. A memory leak is possible. soapysdroutput.cpp 612
V773 Visibility scope of the 'log2Interps' pointer was exited without releasing the memory. A memory leak is possible. soapysdroutput.cpp 612

V764 Possible incorrect order of arguments passed to 'create' function: 'individualGainsChanged' and 'globalGainChanged'. soapysdrinput.cpp 1260

f4exb, 2018-12-03:
If the build of the server is conditional then the condition should be ON by default. It is true that it is not normally needed for Windows since it makes no sense to run headless on Windows.

Then the GUI version should also be optional also with default ON. Because similarly you would not run the GUI version on smallish SBCs like the Raspberry Pi.

The default should be ON for the unwary to compile all versions by default.