f4exb/sdrangel #255: SoapyRemote support

sgoudelis, 2018-11-27:
SoapySDR support is working for me for the most part. It is detecting locally attached SDR but it is not working anymore for SDRs connected to a remote box and using SoapyRemote. Other software that uses SoapyRemote works fine with my installation. Any ideas what to look for ? If you like some log outputs please let me know.

f4exb, 2018-11-27:
I just made it work so it works. You have first to make sure you can make the connection with the SoapySDR utilities. You should get no red lines in the SoapySDRServer nor the SoapySDRUtil outputs. Make sure the udev rules are correctly defined, add the necessary libraries to the LD_LIBRARY_PATH...

On server side: 
```
/opt/install/SoapySDR/bin/SoapySDRServer --bind 192.168.1.47
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.5.1-g5a647db9
Server UUID: 2cc4b240-1857-156b-8567-1c0b007f0101
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server... 
Connecting to DNS-SD daemon... 
[INFO] Avahi version:  avahi 0.6.32
[INFO] Avahi hostname: rpi304
[INFO] Avahi domain:   local
[INFO] Avahi FQDN:     rpi304.local
[INFO] avahi_entry_group_add_service(rpi304._soapy._tcp)
Press Ctrl+C to stop the server
```

On client side:
```
/opt/install/SoapySDR/bin/SoapySDRUtil --find="remote=192.168.1.47"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################
...
Found device 3
  available = Yes
  driver = remote
  label = Generic RTL2832U OEM :: R8202005
  manufacturer = Realtek
  product = RTL2838UHIDIR
  remote = tcp://192.168.1.47:55132
  remote:driver = rtlsdr
  rtl = 0
  serial = R8202005
  tuner = Rafael Micro R820T
```
So basically this is SoapySDR business not SDRangel's

The only thing to notice is that the timeout value on `readStream` is not sufficient when the Rx is remote so there are many of these messages in the log:
```
2018-11-27 22:38:30.294 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-27 22:38:30.302 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-27 22:38:30.309 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-27 22:38:30.316 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
```

sgoudelis, 2018-11-28:
Indeed you were right about a few things. Its all working now. Closing.

closed 2018-11-28 by sgoudelis

synthead, 2021-09-19:
@sgoudelis, how did you get this to work?