f4exb/sdrangel #256: SoapyRemote support: flood of messages

sgoudelis, 2018-11-28:
There is a flood of messages coming out of sdrangel on the console like these:

```
2018-11-28 20:54:13.186 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-28 20:54:13.193 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-28 20:54:13.201 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-28 20:54:13.208 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-28 20:54:13.215 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-28 20:54:13.222 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-28 20:54:13.234 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-28 20:54:13.241 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-28 20:54:13.249 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-28 20:54:13.256 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-28 20:54:13.263 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-28 20:54:13.270 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-28 20:54:13.277 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-28 20:54:13.284 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-28 20:54:13.292 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
2018-11-28 20:54:13.299 (W) SoapySDRInputThread::run: timeout: flags: 0 timeNs: 0 timeoutUs: 6972
```
Any idea what they mean ? This is not necessarily associated with a problem but right now I am getting choppy sound on all modes coming over the lan (1GBps) with an RTL SDR at all sample rates.

f4exb, 2018-11-28:
Already mentioned in #255 
It means we have to use an exaggerated timeout when going through network. The fix is already on dev branch with a minimal timeout of 250ms.

sgoudelis, 2018-11-28:
Cool! Thank you

closed 2018-11-28 by sgoudelis