f4exb/sdrangel #260: 'make install' does not install sdrangel.desktop

probonopd, 2018-12-02:
`make install` does not install `sdrangel.desktop`:

```
$ make DESTDIR=appdir -j$(nproc) install ; find appdir/
appdir/
appdir/usr
appdir/usr/share
appdir/usr/share/sdrangel
appdir/usr/share/sdrangel/udev-rules
appdir/usr/share/sdrangel/udev-rules/fcd.rules
appdir/usr/share/sdrangel/udev-rules/64-limesuite.rules
appdir/usr/share/sdrangel/udev-rules/mirisdr.rules
appdir/usr/share/sdrangel/udev-rules/install.sh
appdir/usr/share/sdrangel/udev-rules/95-perseus.rules
appdir/usr/share/sdrangel/udev-rules/52-airspy.rules
appdir/usr/share/sdrangel/udev-rules/53-hackrf.rules
appdir/usr/share/sdrangel/udev-rules/53-adi-plutosdr-usb.rules
appdir/usr/share/sdrangel/udev-rules/rtl-sdr.rules
appdir/usr/share/sdrangel/udev-rules/52-airspyhf.rules
appdir/usr/share/sdrangel/udev-rules/88-nuand.rules
appdir/usr/bin
appdir/usr/bin/sdrangelbench
appdir/usr/bin/sdrbase.rcc
appdir/usr/bin/sdrangelsrv
appdir/usr/bin/sdrangel
appdir/usr/lib
appdir/usr/lib/libsdrbench.so
appdir/usr/lib/libhttpserver.so
appdir/usr/lib/libswagger.so
appdir/usr/lib/liblogging.so
appdir/usr/lib/pluginssrv
appdir/usr/lib/pluginssrv/channelrx
appdir/usr/lib/pluginssrv/channelrx/libdemodwfmsrv.so
appdir/usr/lib/pluginssrv/channelrx/libdemodamsrv.so
appdir/usr/lib/pluginssrv/channelrx/libdemodnfmsrv.so
appdir/usr/lib/pluginssrv/channelrx/libdemodbfmsrv.so
appdir/usr/lib/pluginssrv/channelrx/libdemodssbsrv.so
appdir/usr/lib/pluginssrv/channelrx/libudpsinksrv.so
appdir/usr/lib/pluginssrv/channeltx
appdir/usr/lib/pluginssrv/channeltx/libudpsourcesrv.so
appdir/usr/lib/pluginssrv/channeltx/libmodssbsrv.so
appdir/usr/lib/pluginssrv/channeltx/libmodatvsrv.so
appdir/usr/lib/pluginssrv/channeltx/libmodnfmsrv.so
appdir/usr/lib/pluginssrv/channeltx/libmodwfmsrv.so
appdir/usr/lib/pluginssrv/channeltx/libmodamsrv.so
appdir/usr/lib/pluginssrv/samplesource
appdir/usr/lib/pluginssrv/samplesource/libinputfilesourcesrv.so
appdir/usr/lib/pluginssrv/samplesource/libinputtestsourcesrv.so
appdir/usr/lib/pluginssrv/samplesink
appdir/usr/lib/pluginssrv/samplesink/liboutputfilesinksrv.so
appdir/usr/lib/libsdrbase.so
appdir/usr/lib/libqrtplib.so
appdir/usr/lib/libsdrgui.so
appdir/usr/lib/plugins
appdir/usr/lib/plugins/channelrx
appdir/usr/lib/plugins/channelrx/libdemodbfm.so
appdir/usr/lib/plugins/channelrx/libudpsink.so
appdir/usr/lib/plugins/channelrx/libchanalyzer.so
appdir/usr/lib/plugins/channelrx/libdemodam.so
appdir/usr/lib/plugins/channelrx/libdemodlora.so
appdir/usr/lib/plugins/channelrx/libdemodatv.so
appdir/usr/lib/plugins/channelrx/libdemodwfm.so
appdir/usr/lib/plugins/channelrx/libdemodssb.so
appdir/usr/lib/plugins/channelrx/libdemodnfm.so
appdir/usr/lib/plugins/channeltx
appdir/usr/lib/plugins/channeltx/libmodam.so
appdir/usr/lib/plugins/channeltx/libmodnfm.so
appdir/usr/lib/plugins/channeltx/libmodssb.so
appdir/usr/lib/plugins/channeltx/libmodwfm.so
appdir/usr/lib/plugins/channeltx/libmodatv.so
appdir/usr/lib/plugins/channeltx/libudpsource.so
appdir/usr/lib/plugins/samplesource
appdir/usr/lib/plugins/samplesource/libinputfilesource.so
appdir/usr/lib/plugins/samplesource/libinputtestsource.so
appdir/usr/lib/plugins/samplesink
appdir/usr/lib/plugins/samplesink/liboutputfilesink.so
appdir/usr/lib/libsdrsrv.so
```

I am missing desktop and icon files.

f4exb, 2018-12-02:
PR please

f4exb, 2018-12-28:
Done in v4.3.2

closed 2018-12-28 by f4exb