f4exb/sdrangel #262: Channel Analyzer: fix rational downsampler

f4exb, 2018-12-08:


GkvJeep, 2018-12-09:
The problem disappears, at the first event from change "lowpass filter cut-off frequency".

f4exb, 2018-12-09:
That's for number 2. True that 1 and 2 are not exactly related. After some trial and error it seems that the interpolator problem (1) lies in the use of SSE2 based code. With modern compilers it might not be needed anyway.

GkvJeep, 2018-12-09:
I do not see anything suspicious, given that the sampling rate is changed 10 times.

Sample 1.28 M/S 
[image]

Sample 10.24 M/S 
[image]

f4exb, 2018-12-09:
for point no 2 it seems the down channelizer rate change is forwarded from the input object to scope and spectrum GUIs while it is also done from the GUI. Both messages conflict and only the one sent from the GUI should be considered as it handles possible rational resampling.

GkvJeep, 2018-12-10:
I do not observe such an effect.
Baseband is at 2.56 MS/s without rational downsampling:
[image]

Rational downsampling at 1.194 MS/s (factor 2.143):
[image]

GkvJeep, 2018-12-10:
I do not understand, why limit to 1.996 (1.0/0.501)
Example. 
Joseph D. Gaeddert
http://liquidsdr.org/doc/msresamp/
Rational downsampling at 8.547 or (1.0/0.117f)
[image]

GkvJeep, 2018-12-10:
I do not understand why limit to 1.996 (1.0/0.501)
Example
Joseph D. Gaeddert
http://liquidsdr.org/doc/msresamp2/
decimate by 2^4=16
[image]

f4exb, 2018-12-11:
The present code for the rational resampler in SDRangel that I have not written myself works like that. Take it as a given. Of course it can be replaced by the Liquid SDR one bit this is not something to be taken lightly. For the moment you should be able to obtain any sampling rate using half band decimators and rational resampler for the last half.

GkvJeep, 2018-12-11:
I agree, read the documentation!  ;-))
Russians have an expression ... 
"Who pays for a girl in a restaurant, he uses it."
Distracted. ((-;
What prevents in the plugin to immediately make a complete decimation, without any additional switching.

f4exb, 2018-12-12:
The channelizer should be configured from the GUI to do just that. Maybe this does not work as expected.

f4exb, 2018-12-28:
Fixed in v4.3.2

closed 2018-12-28 by f4exb