f4exb/sdrangel #264: make  err  on  fedora 28

oystercatcher, 2018-12-21:
I have builds of  SoapySDR and LimeSuite   in  /usr/local      
I tried earlier /opt  but got the same error.    Just thought I would ask about it.   

Looking at the documentation on non-hardware dependencies  are 
they windows only?    I have not included them in the build yet.

Not sure if this is related  http://www.osmocom.org/issues/3729

running make in build dir

[ 37%] Built target sdrbench
[ 37%] Building CXX object devices/limesdr/CMakeFiles/limesdrdevice.dir/devicelimesdr.cpp.o
[ 38%] Building CXX object devices/limesdr/CMakeFiles/limesdrdevice.dir/devicelimesdrparam.cpp.o
/usr/local/src/sdrangel/devices/limesdr/devicelimesdr.cpp: In static member function ‘static bool DeviceLimeSDR::setClockSource(lms_device_t*, bool, uint32_t)’:
/usr/local/src/sdrangel/devices/limesdr/devicelimesdr.cpp:362:49: error: too few arguments to function ‘int LMS_VCTCXORead(lms_device_t*, uint16_t*, bool)’
         if (LMS_VCTCXORead(device, &vcoTrimValue))
                                                 ^
In file included from /usr/local/src/sdrangel/devices/limesdr/devicelimesdr.h:20,
                 from /usr/local/src/sdrangel/devices/limesdr/devicelimesdr.cpp:20:
/usr/local/include/lime/LimeSuite.h:993:26: note: declared here
 API_EXPORT int CALL_CONV LMS_VCTCXORead(lms_device_t * dev, uint16_t *val, bool memory);
                          ^~~~~~~~~~~~~~
/usr/local/src/sdrangel/devices/limesdr/devicelimesdr.cpp:368:49: error: too few arguments to function ‘int LMS_VCTCXOWrite(lms_device_t*, uint16_t, bool)’
         if (LMS_VCTCXOWrite(device, vcoTrimValue))
                                                 ^
In file included from /usr/local/src/sdrangel/devices/limesdr/devicelimesdr.h:20,
                 from /usr/local/src/sdrangel/devices/limesdr/devicelimesdr.cpp:20:
/usr/local/include/lime/LimeSuite.h:982:26: note: declared here
 API_EXPORT int CALL_CONV LMS_VCTCXOWrite(lms_device_t * dev, uint16_t val, bool memoty);
                          ^~~~~~~~~~~~~~~
make[2]: *** [devices/limesdr/CMakeFiles/limesdrdevice.dir/build.make:63: devices/limesdr/CMakeFiles/limesdrdevice.dir/devicelimesdr.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
Scanning dependencies of target plutosdrdevice
[ 38%] Building CXX object devices/plutosdr/CMakeFiles/plutosdrdevice.dir/deviceplutosdr.cpp.o
[ 38%] Building CXX object devices/plutosdr/CMakeFiles/plutosdrdevice.dir/deviceplutosdrbox.cpp.o
make[1]: *** [CMakeFiles/Makefile2:1141: devices/limesdr/CMakeFiles/limesdrdevice.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 39%] Building CXX object devices/plutosdr/CMakeFiles/plutosdrdevice.dir/deviceplutosdrscan.cpp.o
[ 39%] Building CXX object devices/plutosdr/CMakeFiles/plutosdrdevice.dir/deviceplutosdrparams.cpp.o
[ 39%] Building CXX object devices/plutosdr/CMakeFiles/plutosdrdevice.dir/deviceplutosdrshared.cpp.o
[ 39%] Generating ui_basicchannelsettingsdialog.h
[ 40%] Generating ui_commandoutputdialog.h
[ 40%] Building CXX object devices/plutosdr/CMakeFiles/plutosdrdevice.dir/plutosdrdevice_autogen/mocs_compilation.cpp.o
[ 40%] Generating ui_cwkeyergui.h
[ 40%] Generating ui_editcommanddialog.h
[ 40%] Generating ui_externalclockdialog.h
[ 40%] Generating ui_glscopegui.h
[ 40%] Generating ui_glscopemultigui.h
[ 40%] Generating ui_glspectrumgui.h
[ 40%] Generating ui_pluginsdialog.h
[ 40%] Generating ui_audioselectdialog.h
[ 41%] Generating ui_audiodialog.h
[ 41%] Generating ui_samplingdevicecontrol.h
[ 41%] Generating ui_samplingdevicedialog.h
[ 41%] Generating ui_myposdialog.h
[ 41%] Generating ui_transverterdialog.h
[ 41%] Generating ui_loggingdialog.h
[ 41%] Generating ui_discreterangegui.h
[ 41%] Linking CXX shared library libplutosdrdevice.so
[ 42%] Generating ui_intervalrangegui.h
[ 42%] Generating ui_intervalslidergui.h
[ 42%] Generating ui_complexfactorgui.h
[ 42%] Generating ui_arginfogui.h
Scanning dependencies of target sdrgui
[ 42%] Built target plutosdrdevice
[ 42%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/aboutdialog.cpp.o
[ 42%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/mainwindow.cpp.o
[ 42%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/addpresetdialog.cpp.o
[ 43%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/audiodialog.cpp.o
[ 43%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/audioselectdialog.cpp.o
[... 36 lines cut ...]
[ 47%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/tvscreen.cpp.o
[ 48%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/valuedial.cpp.o
[ 48%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/valuedialz.cpp.o
[ 48%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/dsp/scopevis.cpp.o
[ 48%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/dsp/scopevismulti.cpp.o
[ 48%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/dsp/scopevisxy.cpp.o
[ 48%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/dsp/spectrumvis.cpp.o
[ 48%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/dsp/spectrumscopecombovis.cpp.o
[ 48%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/device/deviceuiset.cpp.o
[ 49%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/discreterangegui.cpp.o
[ 49%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/intervalrangegui.cpp.o
[ 49%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/itemsettinggui.cpp.o
[ 49%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/stringrangegui.cpp.o
[ 49%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/dynamicitemsettinggui.cpp.o
[ 49%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/dynamicargsettinggui.cpp.o
[ 49%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/intervalslidergui.cpp.o
[ 49%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/complexfactorgui.cpp.o
[ 50%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/arginfogui.cpp.o
[ 50%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/webapi/webapiadaptergui.cpp.o
[ 50%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/qrc_res.cpp.o
[ 50%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/sdrgui_autogen/mocs_compilation.cpp.o
[ 50%] Linking CXX shared library libsdrgui.so
[ 50%] Built target sdrgui
make: *** [Makefile:130: all] Error 2
[pstan@localhost psbuild]$

f4exb, 2018-12-22:
Contrarily to what is explained in the LimeSuite issue at tag v18.10.0 the function has only two paramters:
`src/lime/LimeSuite.h:API_EXPORT int CALL_CONV LMS_VCTCXOWrite(lms_device_t * dev, uint16_t val);`
Maybe the third parameter was added later. So you should checkout tag v18.10.0 in your LimeSuite clone.

oystercatcher, 2018-12-26:
I restarted the install per the doc and included  the non-hardware dependencies.  I want to thank you for the detailed  doc and the thinking behind the process.   
Although I am a ham, callsign  kf6nam   I only worked packet for a short time in the 80's.   I have learned a lot from this build and am looking forward to using sdrangel with my limesdr mini.

closed 2018-12-26 by oystercatcher