f4exb/sdrangel #271: 4.3.2 -windows 10- loading pluto dll module error

lucapgt, 2019-01-11:
Hi, version 4.3.2 on windows 10 return (on log file) error on loading inputplutosdr.dll module.
On some pc with windows 10 works fine, on other pc with the same windows 10 return the error.
Can you try to fix?
Thanks

f4exb, 2019-01-11:
Probably not with so little information. Moreover if it works on some computers and not on some others may designate a particular issue with the hardware or other issue not directly related to SDRangel. Can you at least give me the log in debug mode?

lucapgt, 2019-01-11:
Ok, how i can start debug mode under Windows?

f4exb, 2019-01-11:
2 ways:

1. From the main top bar: Preferences>Logging a popup opens, choose File log level: Debug, check log to file, click on the folder icon and navigate to where you want the log file to be, give it a name in the "File name" box and click on "Save". Next time you start SDRangel the log will be written to the file.

2. Start SDRangel in a Linux style shell like the one you can obtain when installing Git for Windows (https://git-scm.com/download/win). Once installed click on the "Git bash" application icon and a shell window will appear. Navigate to where sdrangel,exe is with cd /c/... (it uses slashes and not backslashes). Then start it by typing sdrangel.exe. The messages will be logged to the shell window. Make sure you select Console log level: Debug in the Preferences>Logging dialog.

lucapgt, 2019-01-14:
Hi, we found that the problem is related on vc++ version. If we use vc++2017 we have this issue, but when we use vc++ 2010 sdrangel works perfectly with pluto. When you have time, can you investigate on it? Thanks

f4exb, 2019-01-14:
MSVC 2017 is used so it is normal to use the 2017 runtime. Falling back to 2010 looks like a workaround for something else. Note that the libiio that is shipped in the binary bundle is the one from Pothos and I have no clue on which version was used to build it.