f4exb/sdrangel #273: Abort on error from SoapySDRServer

troffasky, 2019-01-11:
When trying to load whilst using RTL-SDR stick a preset which was created whilst using SDRplay RSP1a, sdrangel bails with:

2019-01-11 22:40:46.962 (D) DSPDeviceSourceEngine::handleInputMessages
2019-01-11 22:40:46.962 (D) DSPDeviceSourceEngine::handleInputMessages: message: DSPSignalNotification
2019-01-11 22:40:46.962 (D) DSPDeviceSourceEngine::handleInputMessages: DSPSignalNotification( 2048000 , 1060000 )
2019-01-11 22:40:46.962 (D) DSPDeviceSourceEngine::handleInputMessages: forward message to  SpectrumVis
2019-01-11 22:40:46.962 (D) DSPDeviceSourceEngine::handleInputMessages: forward message to  FileRecord
2019-01-11 22:40:46.962 (D) FileRecord::handleMessage: DSPSignalNotification: m_inputSampleRate:  2048000  m_centerFrequency:  1060000
2019-01-11 22:40:46.962 (D) DSPDeviceSourceEngine::handleSourceMessages: forward message to ThreadedSampleSink( DownChannelizer(AMDemod) )
2019-01-11 22:40:46.962 (D) DownChannelizer::handleMessage: DSPSignalNotification: m_inputSampleRate:  2048000
2019-01-11 22:40:46.962 (D) DownChannelizer::applyConfiguration in= 2048000 , req= 0 , out= 2048000 , fc= 0
2019-01-11 22:40:46.962 (D) DSPDeviceSourceEngine::handleInputMessages: DSPSignalNotification: guiMessageQueue: 0x56085096d598
terminate called after throwing an instance of 'std::runtime_error'
  what():  RemoteError: Invalid IF stage, 1 or 1-6 for E4000
Aborted

I can work around the problem, eventually, by changing the sampling device to FileSource, load the preset, then change sampling device back to the RTL-SDR stick.

I think sdrangel should handle error messages from SoapySDRServer more gracefully. 
Something more bookmark-y which keeps less state than the current presets do probably wouldn't have this problem.

f4exb, 2019-01-12:
I do not understand the last sentence. The serial number of the device is used to store device related settings which allows the same preset to be used on different hardware. However with Soapy this is not possible because the driver name and a sequence number is used as the serial number. I don not want to give better support for SoapySDR anyway.

troffasky, 2019-01-12:
The point of my final sentence was that if the bookmark didn't store the state of the tuner, then loading a bookmark saved with tuner A wouldn't cause sdrangel to fall over when loaded whilst using tuner B.

f4exb, 2019-01-12:
This is how this is supposed to work. So if I get it right the problem occurs when going from RSP1A which can only be supported through SoapySDR (hence the connection to Soapy) to a natively supported RTL-SDR. This is not connected in another way to SoapySDR and by the way there is no "SoapySDRServer" anywhere in the code (you can check).

troffasky, 2019-01-12:
The RSP1a is local through Soapy, the RTL-SDR is remote and accessed through SoapySDRServer.

f4exb, 2019-01-13:
I cannot reproduce the case exactly (may be one needs a E4000 based device). But the device type match issue is visible here:
```
2019-01-13 20:39:28.116 (D) MainWindow::on_presetLoad_clicked
2019-01-13 20:39:28.116 (D) MainWindow::loadPresetSettings: preset [test | test Soapy]
2019-01-13 20:39:28.116 (D) DeviceSourceAPI::loadSourceSettings: Loading preset [test | test Soapy]
2019-01-13 20:39:28.116 (D) Preset::findBestSourceConfig: first of kind matched: id: sdrangel.samplesource.soapysdrinput ser: hackrf-0 seq: 1
2019-01-13 20:39:28.116 (D) DeviceSourceAPI::loadSourceSettings: center frequency: 435000000 Hz
2019-01-13 20:39:28.116 (D) DeviceSourceAPI::loadSourceSettings: deserializing source sdrangel.samplesource.soapysdrinput[3]: remote-1
2019-01-13 20:39:28.116 (D) DeviceUISet::loadRxChannelSettings: Loading preset [test | test Soapy]
2019-01-13 20:39:28.117 (D) DeviceUISet::loadRxChannelSettings: 0 channel(s) in preset
2019-01-13 20:39:28.212 (D) SoapySDRInputGui::updateHardware
2019-01-13 20:39:28.212 (D) SoapySDRInput::handleMessage: MsgConfigureSoapySDRInput
```
Step 1: use HackRF through Soapy and store a preset
Step 2: use RTLSDR (or whatever) through SoapyRemote. The best match falls on the settings for the HackRF because all these devices are matched to the SoapySDR hardware ID. 

Matching the SoapySDR driver key (the first part before "-" like "hackrf" from "hackrf-0") would solve a few compatibility issues.