f4exb/sdrangel #274: XTRX not starting

freqmod, 2019-01-12:
Hi

XTRX does not start aquisition with the internal support, most likely due to some race conditions. The soapy version works.

I have created a patch that makes it start, but it seems configuring xtrx while it is running is still causing problems.

freqmod, 2019-01-12:
`

From 4176bcd4b89339390882cb4a7438c7337f9705ef Mon Sep 17 00:00:00 2001
From: ...
Date: Sat, 12 Jan 2019 11:10:20 +0100
Subject: [PATCH] Make XTRX start with internal plugin

diff --git a/plugins/samplesource/xtrxinput/xtrxinput.cpp b/plugins/samplesource/xtrxinput/xtrxinput.cpp
index bbf2727b0..a7efb7ab1 100644
--- a/plugins/samplesource/xtrxinput/xtrxinput.cpp
+++ b/plugins/samplesource/xtrxinput/xtrxinput.cpp
@@ -347,13 +347,14 @@ bool XTRXInput::start()
     xtrxInputThread->setFifo(requestedChannel, &m_sampleFifo);
     xtrxInputThread->setLog2Decimation(requestedChannel, m_settings.m_log2SoftDecim);
 
+    applySettings(m_settings, true);
+
     if (needsStart)
     {
         qDebug("XTRXInput::start: (re)start thread");
         xtrxInputThread->startWork();
     }
 
-    applySettings(m_settings, true);
 
     qDebug("XTRXInput::start: started");
     m_running = true;
-- 
2.19.1
`

f4exb, 2019-01-12:
I have issued warnings that XTRX is still unstable and you might need to stop and re-start manually.

f4exb, 2019-01-16:
Tried to apply the patch. I did not test it in the long run at least it does not seem to hurt but still does not solve the issue when starting XTRX the first time after plug in:
```
2019-01-16 08:39:02.485 (D) DeviceXTRX::open: serial: usb3380://2/3/16
08:39:02.590302 INFO:   [USB3] USB3380 dual fly GPEP RX mode is on, TX mode is on
08:39:02.590579 INFO:   [3380] USB338x Speed: 480 Mbps
08:39:02.590879 WARN:   [3380] Need to reinitialize the USB device dueto configuration change
08:39:02.712556 ERROR:  [3380] Unable to reset USB3380: error `Entity not found` (-5)!
08:39:02.712626 ERROR:  [USB3] Unable to reinitialize context: error: -13
```
Then closing SDRangel and after all successive starts without unplugging the device the open sequence works fine:
```
2019-01-16 08:40:30.919 (D) DeviceXTRX::open: serial: usb3380://2/3/17
08:40:31.022354 INFO:   [USB3] USB3380 dual fly GPEP RX mode is on, TX mode is on
08:40:31.022644 INFO:   [3380] USB338x Speed: 480 Mbps
08:40:31.032047 INFO:   [3380] Initialized PCIe dev 0100 [10ee:7012]
08:40:31.032357 INFO:   [3380] IO thread started
08:40:31.032484 INFO:   [CTRL] USB3_2/3/17: XTRX Rev4 (04000013)
08:40:31.032642 INFO:   [BPCI] USB3_2/3/17: RX DMA STOP MIMO (BLK:0 TS:0); TX DMA STOP MIMO @0.0
08:40:31.032769 INFO:   [USB3] USB3_2/3/17: Device `usb3380://2/3/17` was opened
CPU Features: SSE2+ SSE4.1+ AVX+ FMA+
08:40:31.148451 INFO:   [CTRL] USB3_2/3/17: RFIC_GPIO 0x000304
08:40:31.248700 INFO:   [CTRL] USB3_2/3/17: FPGA V_GPIO set to 3280mV
08:40:31.248846 INFO:   [CTRL] USB3_2/3/17: LMS PMIC DCDC out set to VA18=1880mV VA14=1480mV VA12=1340mV
08:40:31.252870 INFO:   [CTRL] USB3_2/3/17: FPGA V_IO set to 1800mV
08:40:31.262950 INFO:   [CTRL] USB3_2/3/17: RFIC_GPIO 0x000306
08:40:31.274830 INFO:   [LSM7] USB3_2/3/17: LMS VER:7 REV:1 MASK:1 (3841)
08:40:31.274843 INFO:   [CTRL] USB3_2/3/17: RFIC_GPIO 0x00031e
08:40:31.274978 INFO:   [DBGP] Starting XTRX debug thread
```

f4exb, 2019-01-18:
XTRX is a piece of shit and support will be removed in the next version

f4exb, 2019-01-19:
It seems the latest version of XTRX images libraries do a better job at startup. It is now working the first time after plug in.

f4exb, 2019-01-20:
Also concerning XTRX the detection of GPS lock is causing the program to hang when GPS is not locked. A fix is commited on dev branch and will be part of next release.

freqmod, 2019-01-24:
Tested now, with the newest version of xtrximages and sdrangel. It seems to work without any changes.

f4exb, 2019-01-29:
Ok cool. I have found that my XTRX still hangs sometimes but at least the start and operation works OK globally. So I close this issue.

closed 2019-01-29 by f4exb