f4exb/sdrangel #275: Randomise colour of duplicate demodulators

troffasky, 2019-01-12:
When adding an additional demodulator of the same type, the colour [or maybe just saturation] should be randomised to make it clearer which one is which.

f4exb, 2019-01-12:
You can change the colour yourself. Right clicking on the little grey square on the top left corner of the demodulator window will open a dialog to let you do so.

So no point in changing anything here.

closed 2019-01-12 by f4exb

troffasky, 2019-01-12:
Yes, I am aware that I can change the colour manually. The reason for this feature request is that I believe usability would be improved [faster workflow] if the colours of demodulators could be automatically disambiguated.

f4exb, 2019-01-12:
It is a matter of taste and I don't think so. The color is related to the type of demodulator.

f4exb, 2019-01-13:
Note that this could be scripted externally using the REST API.