f4exb/sdrangel #277: PlutoSDR isn't in the list anymore

beta-tester, 2019-01-16:
Hi,
since SDRangel windows release, sdrangelmw_v4.3.0.7z, there is no PlutoSDR anymore in the list of Sources / Sinks.
ok, in the release notes of v4.3.0 was written, that PlutoSDR wasn't included, but the versions after, there was written, that PlutoSDR is back again.
PlutoSDR is not available anymore, not in sdrangelms_v4.3.1.7z nor in sdrangelms_v4.3.2.7z.
i only have FileSource, SDRdaemonSource, TestSource / FileSink, SDRdaemonSink available there.

... did i missed something?

PS.: i just updated the firmware of the PlutoSDR to [v0.30](https://github.com/analogdevicesinc/plutosdr-fw/releases/). With the old version of SDRangel, sdrangel_v4.2.4.7z, i can still select my PlutoSDR successfully.
i just tried the new versions of SDRangel after the firmware update. befor i updated, i only used the sdrangel_v4.2.4.7z

lucapgt, 2019-01-17:
pluto is present in plugin list of sdrangel?

beta-tester, 2019-01-17:
the files
`D:\SDR\sdrangelms_v4.3.2\sdrangelms\plugins\samplesource\inputplutosdr.dll`
and
`D:\SDR\sdrangelms_v4.3.2\sdrangelms\plugins\samplesink\outputplutosdr.dll`
do exist.
but nothing there to select in the sdrangel application:
[image]

lucapgt, 2019-01-17:
[image]

lucapgt, 2019-01-17:
yes, install microsoft vc++ 2010, after that pluto plugin appear in the list and you can select it in the sampling device.

beta-tester, 2019-01-17:
yes, installing vcredist_x64 for 2010 did the trick...
now i have my PlutoSDR in v4.3.2...
thank you very much!!!

closed 2019-01-17 by beta-tester