f4exb/sdrangel #282: xtrx windrivers installed but not found in sdrangel

DG9BFC, 2019-02-01:
just installed the win drivers ... that was tricky but finally it worked
so ... xtrx found in device manager ...
but it is not found in sdrangel ...
need some help to get it working

DG9BFC, 2019-02-02:
hnmm i do not know how to do that ... i am just a user ... not a programmer
i thought it would work similar to pluto or lime (install drivers and start soft)
if drivers installed and soft supports the hardware it should work that way
why is it not working?
i have no idea how to compile a windows soft ... even installing of the windows drivers was a bit tricky (but i finally found out how to do it) ...
maybe you can help me on that?? ...
sdrangel is the first and only workable sdrsoftware that has the xtrx ... otherwise i have a nice but expensive letterweight for 350 euros

DG9BFC, 2019-02-02:
i found and xtrx/sdrangel fork .... but that stops at version 385 (no version 4 or higher) so that was also a dead end street :-(

f4exb, 2019-02-02:
I am not responsible of this blame the XTRX guys

DG9BFC, 2019-02-02:
that is what i found in xtrx-sdr/images:

Is Windows driver support planned?

We plan to add Windows PCIe drivers but we don't have specific schedule for this right now. If it's blocking factor please contact us xtrx@fairwaves.co. However, all other software layers are windows compatible and USB3380 adapter should work via WinUSB.

f4exb, 2019-02-02:
It's been some time since the last time I visited XTRX Github repo. This was posted 18 days ago: https://github.com/xtrx-sdr/images/tree/master/binaries/Windows/x64

Maybe it can work. No guarantee but I can have a try

f4exb, 2019-02-02:
OK let's keep it open maybe forever...

DG9BFC, 2019-02-02:
hopefully not forever .... i did install the drivers fro here
https://github.com/xtrx-sdr/images/tree/master/binaries/Windows/driver_usb3380
..... i do not know if that were the correct ones but now the xtrx is found in the devices manager
... i also downloaded the drivers from the x64 folder (but have no idea how to intall them) ... would be nice if you could help me on that cause i guess the fairwaves guys can not help me with getting your software to work with the xtrx (or is it still a driver fault?!?)

f4exb, 2019-04-28:
This may never happen. Dropped.

closed 2019-04-28 by f4exb