f4exb/sdrangel #284: Build error

KC7NOA, 2019-02-01:
Probably temporary ....

I use Limesuite from the PPA and the rest ... nothing built but SdrAngel

Built on Rock64 (arm )

Scanning dependencies of target sdrangelbench
[ 50%] Building CXX object CMakeFiles/sdrangelbench.dir/appbench/main.cpp.o
[ 50%] Building CXX object
CMakeFiles/sdrangelbench.dir/sdrangelbench_autogen/mocs_compilation.cpp.o
[ 51%] Linking CXX executable sdrangelbench
[ 51%] Built target sdrangelbench
[ 51%] Automatic MOC for target limesdrdevice
[ 51%] Built target limesdrdevice_autogen
[ 51%] Building CXX object
devices/limesdr/CMakeFiles/limesdrdevice.dir/devicelimesdr.cpp.o
/home/rock64/Downloads/sdrangel/devices/limesdr/devicelimesdr.cpp: In
static member function 'static bool
DeviceLimeSDR::setClockSource(lms_device_t*, bool, uint32_t)':
/home/rock64/Downloads/sdrangel/devices/limesdr/devicelimesdr.cpp:362:49:
error: too few arguments to function 'int
LMS_VCTCXORead(lms_device_t*, uint16_t*, bool)'
         if (LMS_VCTCXORead(device, &vcoTrimValue))
                                                 ^
In file included from
/home/rock64/Downloads/sdrangel/devices/limesdr/devicelimesdr.h:20:0,
                 from
/home/rock64/Downloads/sdrangel/devices/limesdr/devicelimesdr.cpp:20:
/usr/include/lime/LimeSuite.h:993:26: note: declared here
 API_EXPORT int CALL_CONV LMS_VCTCXORead(lms_device_t * dev, uint16_t
*val, bool memory);
                          ^~~~~~~~~~~~~~
/home/rock64/Downloads/sdrangel/devices/limesdr/devicelimesdr.cpp:368:49:
error: too few arguments to function 'int
LMS_VCTCXOWrite(lms_device_t*, uint16_t, bool)'
         if (LMS_VCTCXOWrite(device, vcoTrimValue))
                                                 ^
In file included from
/home/rock64/Downloads/sdrangel/devices/limesdr/devicelimesdr.h:20:0,
                 from
/home/rock64/Downloads/sdrangel/devices/limesdr/devicelimesdr.cpp:20:
/usr/include/lime/LimeSuite.h:982:26: note: declared here
 API_EXPORT int CALL_CONV LMS_VCTCXOWrite(lms_device_t * dev, uint16_t
val, bool memoty);
                          ^~~~~~~~~~~~~~~
devices/limesdr/CMakeFiles/limesdrdevice.dir/build.make:62: recipe for
target 'devices/limesdr/CMakeFiles/limesdrdevice.dir/devicelimesdr.cpp.o'
failed
make[2]: *** [devices/limesdr/CMakeFiles/limesdrdevice.dir/devicelimesdr.cpp.o]
Error 1
CMakeFiles/Makefile2:1137: recipe for target
'devices/limesdr/CMakeFiles/limesdrdevice.dir/all' failed
make[1]: *** [devices/limesdr/CMakeFiles/limesdrdevice.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
rock64@rock64:~/Downloads/sdrangel/build$

f4exb, 2019-02-02:
Obviously this is not the right version of LimeSuite

f4exb, 2019-02-05:
The extra bool parameter has been there for a while but was them removed 11 days ago.

closed 2019-02-07 by f4exb

KC7NOA, 2019-02-08:
Thanks for the help... Lime Suite is from repositories .... so I really don't know how to pull something newer or build with video driver issues