f4exb/sdrangel #285: Open .wav files recorded from SDRSharp

loic2665, 2019-02-03:
Hi,

So I just discovered SDRAngel, that is a beautiful and complete software (Big thanks for DSD decoder built in !)

But I recorded a Raw signal from SDRSharp and it output a `.wav` file.
I tried to open it via SDRAngel (from fileSource option), but no luck i needed a `.sdriq` file, and not a `.wav`

I added a `.sdriq` extention to the file but obviously, this didn't worked. I read about a File input plugin,  from the issue #152  but I have no idea where to enable it (F7 to record !? F8 to stop ?? (still a `.sdriq` file... and no reading option...) 

Could you please guide me a bit ? Or is there a way to convert a `.wav` file to a `.sdriq` file ? Is there metadata to the `.sdriq` file ?

Thanks.

f4exb, 2019-02-04:
Hi,

yes there is metadata in the .sdriq files. You will find a utility to re-create this meta data here: https://github.com/f4exb/sdrangel/tree/master/rescuesdriq

I suppose the .wav files in question are actually I/Q samples saved as audio stereo. You will need to first convert it to raw format (2 channel S16LE) using a utility like `sox`. You may loose some dynamic due to the conversion to 16 bit samples. .sdriq files may also be saved as 24 bit over 32 bit samples (8 zero bit MSB) but I am not sure `sox` can convert to this format.

Of course you must know at least the original sample rate. The other parameters like center frequency and initial timestamp are not critical to operation.

loic2665, 2019-02-07:
Hi, thanks for your time, and sorry for my late answer.

Yes, this is I/Q sample. I'm less sure about the stereo feature.
But yes, I know the sample rate, we can see it through Audacity too. (For stereo channel too)

This seems a bit tricky to open a simple `.wav` file. Maybe to improve?

But I will take a look. And I will give you a feedback.

Thanks.

f4exb, 2019-02-07:
.wav files are not supported. You need the metadata header.

closed 2019-02-07 by f4exb